MCUdude/MightyCore #125: W5100 Ethernet Module Won't Work With ATmega32

parthbhat13, 2018-12-17:
Hi,
Since i have found out your Cores i have always been using them in lot of my projects as i don't like having bootloaders on my controllers, and it helps me to use the other controllers like 644P, 324 and the one  i use the most is Atmega32. Thanks for making this Library.

But, right now i am facing a lot of issues , on the ethernet library, i got the w5100 modules with me and connected it properly with the SPI Connections and even the chip Select, to cross check, i opened your library and checked the code where it shows the BitWise operator pointing to the DDRB, PORTB, PIN4 for ChipSelect. 
Still after a lot and lot of tries i cannot get this working, its just not talking with board, and not showing anything specific with all the examples i ran, Only once it worked and it was a eureka moment for me. but it was very sudden and temporary. 

I hear you asking, Did i see if your library is getting compiled for ethernet and SPI , Yes it is, and i even tried multiple Atmega32's and even Multiple W5100's . But it just won't work. 

i found it working with the 328P, Mega2560.

I hope i was clear with what issue i am having!
Thanks in advance
Parth Temkar

MCUdude, 2018-12-17:
Hi!
I haven't had the time to test with any hardware yet. However can you please post a picture if your setup. According to the official Arduino [ethernet shield schematic](https://www.arduino.cc/en/uploads/Main/arduino-ethernet-shield-06-schematic.pdf) you'll have to use the 2x3 header when connecting the SPI lines to the ATmega32. The pin row (D11-D13) doesn't seem to be connected to anything.

MCUdude, 2018-12-17:
Can you try downgrading to MightyCore v1.0.8?
This version uses a different version of the ethernet library. Would be interesting to know if this version works better for you.

MCUdude, 2018-12-17:
I just connected a typical ethernet shield clone bought from Ebay to an ATmega324P. I connected it sung the six pin port + a wire to digital pin 4 (PB4). It works like a charm and seems to be stable.
I can try using an ATmega32 tomorrow if that's any help.

Are you sure all your wires are intact? I've spent hours of troubleshooting before figuring out that one of my wires was faulty.

parthbhat13, 2018-12-17:
Nope, Still no luck and its still behaving the same way, the WebClient Example wont work at all. The webServer is behaving very funny, at first it will print some very random IP address, Then after resitting it multiple times it will accept the reset and then it will, load the page and that will only work Just once. 
i am attaching the image of how it all looks like. 

and yes i re-checked my wiring multiple times and i even used the same wires and same module with the normal arduino with your minicore, it still works amazing, but wont work with Atmega32 and Atmega16. I did all the troubleshooting since last 12Hrs, Havent even slept the whole Night. The last option which was left with me was to disturb you all on here. i understand how frustrating it gets at times when people ask most silly things. But i tried my best to eliminate all those. 

Here's The screenshot. I will probably but a 644P tomorrow and Check with that once. 
Thanks a lot again!
[image]

MCUdude, 2018-12-18:
Well, I've tested a little more. I uploaded the Webserver sketch to an ATmega16 running at 16 MHz. Everything seems to be working just fine. I can enter its IP address in my web browser and see the website being hosted with the current analogRead values.

I tried uploading the WebClient sketch to the ATmega16, but the sketch is just too big. However my ATmega324P could handle the sketch size, and I don't experience any issues here too.

I don't have a working ATmega32 at the moment because I accidentally zapped my development board with an electric bug zapper a few months ago. I don't think it would give me a different result than the 324P did though.

I believe your issue is hardware related, even though you've basically tried everything. I'm not sure what's causing the problems you're facing, and I'm not able to reproduce it. However from the pictures I can see that the development board you're using does not have bypass capacitors close the the power pins on the AVR microcontroller. This _may_ make the microcontroller unreliable. [In March an issue where opened](https://github.com/MCUdude/MightyCore/issues/72) where I ended up updating the ethernet library to a newer version. The new version was tested by me and @codmpm. Neither of us experienced any issues IIRC. I tested using the W5100 board I'm using right now, and a W5500 based board.

parthbhat13, 2018-12-19:
hi,
Thanks a lot for your reply, well i even shifted everything on a breadboard now, and rechecked the connections like 100 times now, but i still am not able to get it working, and i even changed the modules and all the modules are working. I even changed the Atmega32 , tried it on 5 different Atmega32 but still  i can see nothing is working for me, i just dont understand whats the whole trouble here.

MCUdude, 2018-12-19:
I just tested with at ATmega32. Still working perfectly. I'm using the same kind of development board as I sell, and a cheap knock-off W5100 ethernet shield.

[image]

BTW what clock frequency are your running your ATmega32 on? And do you upload code using a programmer or a USB to serial interface?

parthbhat13, 2018-12-23:
hi,
sorry for late replay, but i was trying to troubleshoot the issues, i'm in the process to make an Atmega644p Development board to test it. my friend actually re-did the wiring with the atmega32 and it started working, i really don't know what wrong i did, as he followed my instructions to do so. then i tried the 644P with a basic breakout and still i faced the similar issue what i faced with the Atmega32 earlier. i fee l its the improper breadboard connections so i am making a development board.  i will update you with what exactly is going wrong. 
and my setup was: Atmega32 programmed with usbAsp and was running at 16Mhz. 
 i feel the crystal is the issue somewhere, as if i try to run anything on internal oscillator it would run atleast once. but ill keep you updated with the process. 
Thanks a lot for help!

MCUdude, 2018-12-23:
> my friend actually re-did the wiring with the atmega32 and it started working

It's not easy to figure out what's wrong with your setup, but atleast you've proven that it *is* possible to get it working with an ATmega32. It's probably not a MightyCore related issue, so I'm closing this issue.

Thanks for using MightyCore 🙂

closed 2018-12-23 by MCUdude