MCUdude/MightyCore #127: arduino-cli Support

weswitt, 2019-01-04:
I know that arduino-cli is very new and is still an "alpha preview", but it would be awesome of there was mightycore support for arduino-cli. Have you considered this?

https://github.com/arduino/arduino-cli

per1234, 2019-01-04:
What makes you think it's not already supported?

weswitt, 2019-01-04:
Maybe I just assumed it wasn't. Is it supported?

weswitt, 2019-01-04:
One thing I've seen is that after downloading ardiuno-cli the first time you must install board support for the boards that you want to use. For example if you are programming an UNO then you need to do "arduino-cli core install arduino:avr". Likewise you need to install the board package for ESP8266 (I'm using this too). So it seems that there would have to be a mightycore board package for arduino-cli too otherwise the arduino-cli build system would know nothing about the 1284p core, for example.

weswitt, 2019-01-04:
In fact if you run "arduino-cli board listall" you'll find that the 1284p is not in the list of supported boards.

per1234, 2019-01-04:
There is no such thing as a "board package for arduino-cli". Of course Arduino made arduino-cli work with standard Arduino board packages. arduino-cli will soon be used by the Arduino IDE. If it works with the Arduino IDE, it works with arduino-cli. If not, that's a bug with arduino-cli, not with MightyCore.

weswitt, 2019-01-04:
Actually there is such a thing. See this: https://github.com/arduino/arduino-cli/issues/44

The arduino-cli config file(s) reference a package json file that describes the board support for the arduino-cli system. In that json file it references a package zip file that contains the core files for that board.

Given that there is no arduino-cli package that is compatible with the above described system how do you expect this to work?

per1234, 2019-01-04:
The process is the same as what you did for ESP8266:
https://github.com/arduino/arduino-cli#adding-3rd-party-cores
There is one exception: MightyCore has a dependency on the Arduino AVR Boards toolchain. With the Arduino IDE, you would never notice this since the IDE comes with Arduino AVR Boards bundled so you will always have it installed. However, arduino-cli does not have any hardware package bundled so you need to install both the arduino:avr as well as the MightyCore:avr packages:

In the folder where you have arduino-cli installed create a file named .cli-config.yml
Open .cli-config.yml in a text editor and add the following:
```
board_manager:
  additional_urls:
    - https://mcudude.github.io/MightyCore/package_MCUdude_MightyCore_index.json
```
Save the file

Now you can install Arduino AVR Boards and MightyCore:
```
arduino-cli core update-index
arduino-cli core install arduino:avr
arduino-cli core install MightyCore:avr
```
Now you can see the MightyCore 1284 board is indeed installed:
```
Board Name                              FQBN
ATmega1284                              MightyCore:avr:1284
ATmega16                                MightyCore:avr:16
ATmega164                               MightyCore:avr:164
ATmega32                                MightyCore:avr:32
ATmega324                               MightyCore:avr:324
ATmega644                               MightyCore:avr:644
ATmega8535                              MightyCore:avr:8535
Adafruit Circuit Playground             arduino:avr:circuitplay32u4cat
Arduino BT                              arduino:avr:bt
Arduino Duemilanove or Diecimila        arduino:avr:diecimila
Arduino Esplora                         arduino:avr:esplora
Arduino Ethernet                        arduino:avr:ethernet
Arduino Fio                             arduino:avr:fio
Arduino Gemma                           arduino:avr:gemma
Arduino Industrial 101                  arduino:avr:chiwawa
Arduino Leonardo                        arduino:avr:leonardo
Arduino Leonardo ETH                    arduino:avr:leonardoeth
Arduino Mega ADK                        arduino:avr:megaADK
Arduino Mini                            arduino:avr:mini
Arduino NG or older                     arduino:avr:atmegang
Arduino Nano                            arduino:avr:nano
Arduino Pro or Pro Mini                 arduino:avr:pro
Arduino Robot Control                   arduino:avr:robotControl
Arduino Robot Motor                     arduino:avr:robotMotor
Arduino Uno WiFi                        arduino:avr:unowifi
Arduino Yún                             arduino:avr:yun
Arduino Yún Mini                        arduino:avr:yunmini
Arduino/Genuino Mega or Mega 2560       arduino:avr:mega
Arduino/Genuino Micro                   arduino:avr:micro
Arduino/Genuino Uno                     arduino:avr:uno
LilyPad Arduino                         arduino:avr:lilypad
LilyPad Arduino USB                     arduino:avr:LilyPadUSB
Linino One                              arduino:avr:one
```
Compilation works:
```
arduino-cli compile --fqbn MightyCore:avr:1284:pinout=standard,variant=modelP,BOD=2v7,LTO=Os,clock=16MHz_external Blink
Build options changed, rebuilding all
Sketch uses 1214 bytes (0%) of program storage space. Maximum is 130048 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 16375 bytes for local variables. Maximum is 16384 bytes.
```
Unfortunately, upload does not:
```
arduino-cli upload -p COM9 --fqbn MightyCore:avr:1284:pinout=standard,variant=modelP,BOD=2v7,LTO=Os,clock=16MHz_external Blink
Upload tool 'avrdude' not found.
```
This is a bug in arduino-cli. It seems it does not yet support referencing upload tools from another package. This appears to already have a fix (https://github.com/arduino/arduino-cli/pull/103), but it has not yet been merged.

MightyCore could get around this by providing its own avrdude tool but that is too much trouble to go through just to work around a bug in the alpha arduino-cli which will likely soon be fixed.

>Actually there is such a thing. See this: arduino/arduino-cli#44

That's completely unrelated

> The arduino-cli config file(s) reference a package json file that describes the board support for the arduino-cli system. In that json file it references a package zip file that contains the core [...]

Again, there is no such thing as an "arduino-cli package". There is an Arduino Boards Manager package, [which MightyCore already has](https://github.com/MCUdude/MightyCore#boards-manager-installation). See my above instructions for how to use it.

As for whether arduino-cli currently works with hardware packages installed to the sketchbook folder, I don't have time to check right now. If arduino-cli doesn't have that feature now, then it absolutely will by the time it reaches the production state. Almost certainly the same upload tool bug will apply there as well so it's a waste of time for me to investigate.

weswitt, 2019-01-04:
Awesome -- thanks a lot.

closed 2019-01-04 by weswitt

per1234, 2019-01-05:
I tried out the test build available in https://github.com/arduino/arduino-cli/pull/103 and with that version of arduino-cli, uploading to MightyCore boards works fine.

I was also able to successfully use MightyCore manually installed to the hardware folder of the sketchbook (instead of using the Boards Manager installation method. That can be useful if you want to do beta testing or customization of MightyCore, since only the release versions are available via Boards Manager. If you want to set a custom sketchbook location, you can do that in .cli-config.yml, as shown here:
https://github.com/arduino/arduino-cli/blob/master/.cli-config_example.yml

So it looks like all aspects of arduino-cli will be working perfectly with MightyCore (and presumably the other MCUdude cores) as soon as https://github.com/arduino/arduino-cli/pull/103 is merged. Until then, the test build should work fine.

rnbokade, 2020-06-18:
@per1234 can you please tell what to  put in LTO= instead of Os, so as to enable the LTO

per1234, 2020-06-18:
The [custom board option](https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options) ID is `Os_flto`. This is defined in boards.txt:
https://github.com/MCUdude/MightyCore/blob/v2.0.5/avr/boards.txt#L108
```
1284.menu.LTO.Os_flto=LTO enabled
```
So, for example, if you were using MightyCore's ATmega1284 board, the FQBN with all default options except for enabling LTO would be `MightyCore:avr:1284:LTO=Os_flto`.