MCUdude/MightyCore #128: abort() method problem

migila, 2019-01-22:
Hi, 

I use MightyCore in my project based on atmega644 MCU. Now I need to use circular buffer and I've found this one https://github.com/rlogiacco/CircularBuffer library. But In this library is called abort() method in some cases and Arduino IDE gives me this error

> > In file included from D:\Migi\Elektro\Arduino\portable\sketchbook\libraries\CircularBuffer/CircularBuffer.h:143:0,
> 
>                  from sketch\Foo.h:4,
> 
>                  from sketch\Foo.cpp:1:
> 
> D:\Migi\Elektro\Arduino\portable\sketchbook\libraries\CircularBuffer/CircularBuffer.tpp: In member function 'T CircularBuffer<T, S, IT>::shift()':
> 
> D:\Migi\Elektro\Arduino\portable\sketchbook\libraries\CircularBuffer/CircularBuffer.tpp:64:24: error: there are no arguments to 'abort' that depend on a template parameter, so a declaration of 'abort' must be available [-fpermissive]

  if (count <= 0) abort();

I've done some research and abort() method is defined in stdlib.h header file. I can find this file on two paths:
1) D:\Migi\Elektro\Arduino\hardware\tools\avr\avr\include
2) D:\Migi\Elektro\Arduino\portable\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2\avr\include

All MightyCore files are located here
D:\Migi\Elektro\Arduino\portable\packages\MightyCore\hardware\avr\2.0.0
but there is no file with this name or containing this method.

I've tried copy stdlib.h file to path mentioned above, but it didn't help.  

When I change Arduino board to Arduino Uno everything work fine, but when I change it back to MightyCore ATmega644 compiler gives me this error.

What can I do with this problem?
Thank you

Here is sample project with mentioned problem:
[uno_class.zip](https://github.com/MCUdude/MightyCore/files/2782647/uno_class.zip)

MCUdude, 2019-01-22:
Well, the compiler gives me this warning when compiling for Arduino UNO.

```
In file included from C:\Users\user\Documents\Arduino\libraries\CircularBuffer/CircularBuffer.h:143:0,

                 from C:\Users\user\AppData\Local\Temp\arduino_build_720106\sketch\Foo.h:4,

                 from C:\Users\user\AppData\Local\Temp\arduino_build_720106\sketch\Foo.cpp:1:

C:\Users\user\Documents\Arduino\libraries\CircularBuffer/CircularBuffer.tpp: In member function 'T CircularBuffer<T, S, IT>::shift()':

C:\Users\user\Documents\Arduino\libraries\CircularBuffer/CircularBuffer.tpp:64:24: warning: there are no arguments to 'abort' that depend on a template parameter, so a declaration of 'abort' must be available [-fpermissive]

  if (count <= 0) abort();

                        ^

C:\Users\user\Documents\Arduino\libraries\CircularBuffer/CircularBuffer.tpp: In member function 'T CircularBuffer<T, S, IT>::pop()':

C:\Users\user\Documents\Arduino\libraries\CircularBuffer/CircularBuffer.tpp:75:24: warning: there are no arguments to 'abort' that depend on a template parameter, so a declaration of 'abort' must be available [-fpermissive]

  if (count <= 0) abort();

                        ^
```

The reason for the error is the lack of the -fpermissive flag in the MightyCore source files. IMO adding -fpermissive to the official Arduino boards was a mistake. It hides bugs by converting some errors to warnings. Do yourself a favour and turn on compiler warnings in the IDE settings.

Bottom line: You'd have to fix your code to get rid of this warning/error. YOu should rather ask the author of the library rather than me.

migila, 2019-01-22:
Thank you Hans,

your answer helped me. Adding -fpermissive flag to platform.txt helped me, but anyway I will ask author of that library if there is reason to use abort() method.

closed 2019-01-22 by MCUdude