MCUdude/MightyCore #131: Implemenation at new ATMEGA4809 and ATMEGA3209?

hugopilot, 2019-02-10:
Tiltle says it all basically.

Would it be easy to port?

Also, I really appreciate your work and I use it without problems!

That being said, some interesting links for the 4809/3809:
https://www.microchip.com/wwwproducts/en/ATMEGA4809
https://www.microchip.com/wwwproducts/en/ATMEGA3209 
http://ww1.microchip.com/downloads/en/DeviceDoc/40002016A.pdf

MCUdude, 2019-02-10:
Hi!

The new ATmega3208/4208 and ATmega3209/4209 are completely different from previous AVR families. It would take much time and effort to create a standalone hardware package like MightyCore is. I might do it some time, but right now I don't have time to do it. I don't have any hardware either.

You'll need a programmer that supports the new UDPI interface. I think it's implemented in Avrdude.

MCUdude, 2019-02-10:
However, if I someday decide to add support for them it would probably be a separate Arduino core just for this new family.

closed 2019-02-10 by MCUdude

hugopilot, 2019-02-21:
I just bought a 4809, I will try to make my own. I will keep u posted!

MCUdude, 2019-02-21:
I'm currently working on a new Arduino core that will hopefully support the new ATtinys as well.

I got proper test hardware for ATmega3208, ATmega4808, ATmega3209 and ATmega4809. The core is far from ready, and there is still much to figure out. I will probably go public in a few months. Anyways here's a reference pinout I made yesterday. I will provide an alternative pinout as well, because the Uno Wifi pinout is quite a mess IMO.
[image]

MCUdude, 2019-02-21:
But please do keep me updated! If this turns out to be a task you're able to handle (I don't know your experience with tricky firmware and hardware) I might make the repo public a little sooner. Right now There's really nothing to see. The only thing that's properly implemented are the options in the tools menu (BOD, Clock) and the possibility to use other EDBG chips for uploading. I don't own a UNO Wifi, but I'm using the curiosity boards instead with a different EDBG chip so this was top priority.