MCUdude/MightyCore #133: SPI.h and Wire.h occupy twice as much RAM on 324PB than on other AVRs

MCUdude, 2019-03-13:
ATmega324PB has two i2c and SPI ports. I've made it so that Wire1.h and SPI1.h is automatically included when Wire.h or SPI.h is included (because it's convenient). A downside of this is the memory usage.

Unlike the (included) HardwareSerial library, Wire and SPI occupies memory even though their begin() routines isn't even called. I think the large memory usage is because all buffers and variables is included by the compiler.

Does anybody know how we can prevent the SPI and Wire libraries to be included by the compiler when their begin methods isn't called. At least this is how HardwareSerial.h works.

The outcome of this issue will also affect how these libraries work with ATmega328PB as well.

``` c
#include <Wire.h>

void setup() { }

void loop() { }
```

| Target (LTO disabled) | Blank sketch                    | Blank sketch with Wire.h included  |
|-----------------------|---------------------------------|------------------------------------|
| ATmega324P            | Flash: 470 bytes / RAM: 9 bytes | Flash: 1540 bytes / RAM: 216 bytes |
| ATmega324PB           | Flash: 602 bytes / RAM: 9 bytes | Flash: 2686 bytes / RAM: 423 bytes |

Floessie, 2019-03-14:
@matthijskooijman made that happen for [HardwareSerial](https://github.com/arduino/Arduino/pull/1711) some years ago. Maybe he can provide some hints. Especially the ISR in `twi.c` pulls in a lot of code.

MCUdude, 2019-03-18:
@matthijskooijman would you like to point me in the right direction on how this should be done? I've looked at the HardwareSerial code, but I'm not quite sure what exactly changed HarwareSerial's behavior in your commit.

matthijskooijman, 2019-04-03:
@MCUdude, for HWSerial, the problem was that there is an ISR for each UART. Since ISRs are called by the CPU, they are marked as "used" by the `ISR` macro, causing them to be exempt from the usual "is this code even used" optimizations by gcc. Since the ISR references the global object and the contained buffers, these were also always included.

To work around that, a trick was used. When a .o file is directly specified on the linker commandline, it is included in the link and can only be removed again when it is unused (but not ISRs, because of the "used" attribute). However, when the .o file is included in an .a archive file and then included on the link commandline, the .o file is only included in the link when it is actually needed (i.e. a missing symbol can be resolved by including the .o file). When this happens, the *entire* .o file is included in the link, not just the needed symbol.

Using this trick, you can make sure that everything related to Serial is in one .c/.o file, everything releated to Serial1 is in another, etc. Now, when nothing references Serial1, the .o file for it is not included in the build. But if anything does reference Serial1, the entire .o file is included, including the ISR (even though nothing ever references the ISR directly).

For this to work, linking must happen through an .a file. By default, this happens for the core, but libraries are linked by putting all their .o files on the commandline. However, inside library.properties, you can specify an option (something like dot_a_linkage or something, I think it should be documented) to make the link happen through an .a file.

I hope this helps you understand the serial case. Hopefully the same can be applied to the Wire/SPI libraries here. If you have further questions, feel free to ping me :-)

MCUdude, 2019-04-04:
Adding `dot_a_linkage=true` certainly did the trick! Thank you for pointing that out. I've tested it, and I'm closing this issue with confidence 🎉

closed 2019-04-04 by MCUdude

matthijskooijman, 2019-04-04:
Awesome!

commit 00d18db8 (2019-04-04): Add dot_a_linkage
This prevents these ISR based libraries from occupying space when included but not used. Closes #133