MCUdude/MightyCore #134: Atmega324PB defines

ericlangel, 2019-03-20:
I have Problems with compiling with Atmega324PB

I get Errors that UDRE0, RXC0, U2X0....and so on are not declared. It would be nice to have a transformation with defines for the Atmega324PB.

#define UDRE0 URDE
#define RXC0   RXC
#define ....
...and so on

Is that possible?

MCUdude, 2019-03-20:
I've already added some [support macros for SPI and I2C](https://github.com/MCUdude/MightyCore/blob/master/avr/variants/standard/pins_arduino.h), but not for UART. Give me a few days 🙂

MCUdude, 2019-03-22:
Please give it a try now. I've added all (?) missing macros for the three UARTS.

ericlangel, 2019-03-25:
yes, its working.

Thank you, MCUdude ;)

closed 2019-03-25 by ericlangel

MCUdude, 2019-03-25:
Great to hear it solved your issue. And thanks for reporting!