MCUdude/MightyCore #136: Serial upload problem

prampec, 2019-04-30:
I have uploaded the boot loader from the Arduino IDE, and a sample program as well with a Programmer provided by MightyCore (using external 8Mhz clock). Everything was fine.
But I cannot upload using serial.
I have traced the communication and it seams like the boot loader doesn't work at all. After reset the code starts immediately instead of waiting for serial upload.
(In the image attached you can see, that 100ms after reset, my program code starts up by sending a serial message (yellow vertical line) instead of waiting for upload request (blue vertical lines).)
Are the fuse bits are set correctly by firmware upload?
Any help is welcomed, thanks!
[image]

jandrassy, 2019-04-30:
if you uploaded a sample program over ISP after the bootloader, then you deleted the bootloader

prampec, 2019-04-30:
Oops... okay. Confirmed, that this was the problem. But I'm a bit confused. I thought the bootloader is in the very end of the EEPROM.

closed 2019-04-30 by prampec

jandrassy, 2019-05-01:
the bootloader is at the end of the **flash**.
avrdude sends a command to delete the flash completely