MCUdude/MightyCore #137: On software reset it gets into an infinite bootloop with no way to get out of besides power cycle

LeFEAR, 2019-05-12:
Dear all , 

My setup = atmel ATMEGA1284-PU
not using a bootloader because I had some issues uploading to the micro without Arduino ISP ,(via RX+TX pins )  regardless in my code I used to use this piece of code that was software resetting the CPU 
"
#include <avr/wdt.h>
void softwareReset( uint8_t prescaller) {
  // start watchdog with the provided prescaller
  wdt_enable( prescaller);
  // wait for the prescaller time to expire
  // without sending the reset signal by using
  // the wdt_reset() method
  while(1) {}
}
"
While executing softwareReset the CPU goes into a frenzy and is unrecoverable unless you remove power 

I have no idea what I am doing wrong , 

Any help would be great 
thank you for your time and effort

jandrassy, 2019-05-12:
the wdt is not turned off on reset. you must clear it. bootloaders do it

LeFEAR, 2019-05-12:
So if I program it with the bootloader the issue will go way ? 
I've tried to use the arduino without microcontroller version(just use RX,TX and REST) to upload the sketch but to no avail , 

I will buy a proper programer and go from there and see how it goes ,, 

Thank you for your time and effort

jandrassy, 2019-05-12:
I don't understand what you do, but if you can upload a sketch without bootloader, you can upload a bootloader the same way

LeFEAR, 2019-05-12:
> I don't understand what you do, but if you can upload a sketch without bootloader, you can upload a bootloader the same way

Dear Sir , 
Yes I upload the bootloader with arduino ISP , and it works ( the led is blinking as it should) but
if I try to upload the sketch  including the bootloader (first the bootloader and then use RX,TX pins) to upload the code if fails and the only way to upload code to the CPU is via erasing the bootloader

Anyway thanks I will try to fix this and report back with results 
Thanks

jandrassy, 2019-05-12:
do the settings in Tools menu match your hardware?

LeFEAR, 2019-05-12:
they do , because the bootloader was "burned" successfully But I will get a proper programer and not do it via arduino hardware in the upcoming days

jandrassy, 2019-05-12:
and what is your not 'proper programmer'? if avrdude reports burned then you can't do anything better

LeFEAR, 2019-05-12:
> and what is your not 'proper programmer'? if avrdude r

as I said before  I use Arduino ISP see below how I burned my bootloader the first time 
https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard

after burning the Bootloader I move to this part of the guide "Uploading Using an Arduino Board"
that fails catastrophically 

Therefore I use the first configuration and upload my code by holding down the shift key ( therefore overwriting the bootloader ) 

so I am ditching Arduino ISP and buying an "AVRISP mkII" that should be better suited for the job  (i guess ) 

any further input would be grateful thanks

jandrassy, 2019-05-12:
better buy and USB-to-TTL-Serial adapter. if you want to use the USB-to-TTL-Serial on Uno board, you don't need to remove the MCU. a sketch without Serial.begin is good. and wire RX of Uno to RX (and TX to TX) of the MCU on the breadboard. and groud must be wired (you write "just use RX,TX and REST")

LeFEAR, 2019-05-12:
> better buy an USB-to-TTL-Serial adapter. if you want to use the USB-to-TTL-Serial on Uno board, you don't need to remove the MCU. a sketch without Serial.begin is good. and wire RX of Uno to RX of the MCU on the breadboard (and TX to TX). And ground must be wired (you write "just use RX,TX and REST")

5V and GND are standards somehow I need to power the CPU to be programed , non the less I will see how I will manage to set it up with the USB to TTL ( i have a serial one I am expecting on a USB one to come lastly what settings have to be used in the tools in order to work with the simple USB to TTL serial converter ? ) thank you in advance

hassin23ayz, 2019-06-16:
without bootloader i have inserted this init code at my .ino sketch before void setup() 

uint8_t mcusr_mirror __attribute__ ((section (".noinit")));

void get_mcusr(void) \
  __attribute__((naked)) \
  __attribute__((section(".init3")));
void get_mcusr(void)
{
  mcusr_mirror = MCUSR;
  MCUSR = 0;
  cli();
  wdt_disable();
  sei();
}

this disables watchdog and prevents boot loop 

But i am not sure if this is all that is needed or not

LeFEAR, 2019-07-06:
Dear all , 

sorry for the late reply,

managed to put the bootloader , managed to use a serial adapter to upload the sketch , 

is issue with the freeze on the watchdog still exists

thank you in advance

MCUdude, 2023-07-25:
Optiboot will soon be replaced by Urboot, and I will not spend any time trying to fix this

closed 2023-07-25 by MCUdude