MCUdude/MightyCore #139: TWI1_vect enabled on atmega1284p board -> compile failed

netbomo, 2019-05-28:
Hi,
I've find your new PlatformIO.md.
So I'm updating an old project with the nex platformio.ini file :+1: 
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

; ENVIRONMENT SETTINGS
[env:MightyCore]
platform = atmelavr
framework = arduino

; TARGET SETTINGS
; PlatformIO requires the board parameter. Must match your actual hardware
board = mightycore1284
; Actual target
board_build.mcu = atmega1284p
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L

; BUILD OPTIONS
; Current pinout
board_build.variant = mightycore_bobuino
; Comment out to enable LTO (this line unflags it)
build_unflags = -flto
; Extra build flags
build_flags =

; UPLOAD SETTINGS
; Upload serial port is automatically detected by default. Override by uncommenting the line below
;upload_port = /dev/cu.usbserial*
; Upload baud rate
board_upload.speed = 115200

; Upload using programmer
;upload_protocol = usbasp
; Aditional upload flags
;upload_flags = -Pusb

; SERIAL MONITOR OPTIONS
; Monitor and upload port is the same by default
;monitor_port =
; Serial monitor baud rate
monitor_speed = 9600

framework = arduino
lib_deps =
    https://gitlab.com/gilou_/Rtc_Pcf8563.git

```

But now, when I try to compile, I can see the Wire1.h and SPI1 lib in the dependency graph : 
```

Dependency Graph
|-- <Rtc_Pcf8563> #b998b65
|   |-- <Wire> 1.0
|   |   |-- <Wire1> 1.0
|-- <Windvane> #2cfbf26
|   |-- <commons>
|   |   |-- <ADC_SPI>
|   |   |   |-- <SPI> 1.0
|   |   |   |   |-- <SPI1> 1.0
|-- <SD (MightyCore)> 1.0.7
|   |-- <SPI> 1.0
|   |   |-- <SPI1> 1.0
|-- <Power> #7c73ef2
|   |-- <commons>
|   |   |-- <ADC_SPI>
|   |   |   |-- <SPI> 1.0
|   |   |   |   |-- <SPI1> 1.0
|-- <commons>
|   |-- <ADC_SPI>
|   |   |-- <SPI> 1.0
|   |   |   |-- <SPI1> 1.0
|-- <Anemometer> #81701a8
```

But it's wired because my mcu is defined as atmega1284p and the TWI_vect is normally availlable in iom328pb.h/io55702m322.h/iom324pb.h

as : 
```
/home/gilou/.platformio/packages/framework-arduinoavr/libraries/__cores__/MightyCore/Wire1/src/utility/twi1.c:383:ISR(TWI1_vect)
/home/gilou/.platformio/packages/framework-arduinoavr/libraries/__cores__/MightyCore/Wire/src/Wire.h:30:#if defined(TWI1_vect)
/home/gilou/.platformio/packages/toolchain-atmelavr/avr/include/avr/iom328pb.h:1093:#define TWI1_vect            _VECTOR(40)
/home/gilou/.platformio/packages/toolchain-atmelavr/avr/include/avr/ioa5702m322.h:2534:#define TWI1_vect            _VECTOR(48)
/home/gilou/.platformio/packages/toolchain-atmelavr/avr/include/avr/iom324pb.h:1183:#define TWI1_vect            _VECTOR(42)
```

Have you an idea? 
I can't find a way to track were is declared in my tree dependencies program

Regard,
gilou

MCUdude, 2019-05-28:
[It's a known issue](https://github.com/platformio/platform-atmelavr/issues/141). The root of the problem is the Wire library. If you need to use PlatformIO right now I suggest you replace the Wire library with the one from this repo. Or, you can wait for a new MightyCore release (a week or two maybe).

MCUdude, 2019-05-28:
Oh, you also need to replace the SPI and SPI1 libraries.

netbomo, 2019-05-28:
Ok thank you for the reply. 
I've just comment the lines in the Wire and SPI libs!

closed 2019-05-28 by netbomo