MCUdude/MightyCore #140: SD Library (1.0.7) doesn't compile on ATMega1284P

Emaxoso, 2019-06-03:
Hi, I'm using PlatformIO with MightyCore on an ATMega1284P.
When I try to compile a simple program like this:

```
#include <Arduino.h>
#include <SD.h>
void setup() {}
void loop() {}
```
The compiler gives me compilation errors:
```
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/mightycore1284.html
PLATFORM: Atmel AVR > MightyCore ATmega1284
HARDWARE: ATMEGA1284P 16MHz 16KB RAM (127KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 16 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SD (MightyCore)> 1.0.7
|   |-- <SPI> 1.0
|   |   |-- <SPI1> 1.0
Compiling .pioenvs\mightycore1284\src\main.cpp.o
Compiling .pioenvs\mightycore1284\lib0c9\SPI1\SPI1.cpp.o
In file included from .platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.cpp:14:0:
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h: In static member function 'static void SPI1Class::beginTransaction(SPI1Settings)':
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:203:5: error: 'SPCR1' was not declared in this scope
SPCR1 = settings.spcr1;
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:204:5: error: 'SPSR1' was not declared in this scope
SPSR1 = settings.spsr1;
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h: In static member function 'static uint8_t SPI1Class::transfer(uint8_t)':
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:209:11: error: lvalue required as left operand of assignment
SPDR1 = data;
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:217:14: error: 'SPSR1' was not declared in this scope
while (!(SPSR1 & _BV(SPIF))) ; // wait
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h: In static member function 'static uint16_t SPI1Class::transfer16(uint16_t)':
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:223:11: error: 'SPCR1' was not declared in this scope
if (!(SPCR1 & _BV(DORD))) {
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:224:13: error: lvalue required as left operand of assignment
SPDR1 = in.msb;
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:226:16: error: 'SPSR1' was not declared in this scope
while (!(SPSR1 & _BV(SPIF))) ;
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:228:13: error: lvalue required as left operand of assignment
SPDR1 = in.lsb;
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:230:16: error: 'SPSR1' was not declared in this scope
while (!(SPSR1 & _BV(SPIF))) ;
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:233:13: error: lvalue required as left operand of assignment
SPDR1 = in.lsb;
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:235:16: error: 'SPSR1' was not declared in this scope
while (!(SPSR1 & _BV(SPIF))) ;
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:237:13: error: lvalue required as left operand of assignment
SPDR1 = in.msb;
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:239:16: error: 'SPSR1' was not declared in this scope
while (!(SPSR1 & _BV(SPIF))) ;
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h: In static member function 'static void SPI1Class::transfer(void*, size_t)':
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.h:247:11: error: lvalue required as left operand of assignment
SPDR1 = *p;
[... 51 lines cut ...]
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.cpp:39:20: error: 'SS1' was not declared in this scope
digitalWrite(SS1, HIGH);
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.cpp:45:13: error: 'SS1' was not declared in this scope
pinMode(SS1, OUTPUT);
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.cpp:50:5: error: 'SPCR1' was not declared in this scope
SPCR1 |= _BV(MSTR);
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.cpp:59:13: error: 'SCK1' was not declared in this scope
pinMode(SCK1, OUTPUT);
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.cpp:60:13: error: 'MOSI1' was not declared in this scope
pinMode(MOSI1, OUTPUT);
^
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.cpp: In static member function 'static void SPI1Class::end()':
.platformio\packages\framework-arduinoavr\libraries\__cores__\MightyCore\SPI1\src\SPI1.cpp:74:5: error: 'SPCR1' was not declared in this scope
SPCR1 &= ~_BV(SPE);
^
*** [.pioenvs\mightycore1284\lib0c9\SPI1\SPI1.cpp.o] Error 1

```

Am I doing something wrong or is the library that doesn't compile?

MCUdude, 2019-06-04:
I'm very aware of it. There will be a new release soon that will fix these issues. Meanwhile, you can replace the SPI, SPI1, Wire and Wire1 libraries with the ones from this repo.

Emaxoso, 2019-06-14:
Hi, even replacing the libraries with the ones in this repo cause the same problem.

Emaxoso, 2019-07-24:
Any update on this?

MCUdude, 2019-08-16:
Sorry for the late reply. I accidentally stumbled across the same problem. I'll have to do a new release of MightyCore to fix this, but for now, you can tweak your platformio.ini settings. Replace your `board` parameter with this:

```ini
; TARGET SETTINGS
; PlatformIO requires the board parameter. Must match your actual hardware
board = mightycore1284
; Actual target 
board_build.mcu = atmega1284p
```

ivankravets, 2019-08-19:
See https://github.com/MCUdude/MightyCore/pull/158

A temporary solution is to add `lib_ldf_mode = chain+` to `platformio.ini`. For example,
```
[env:ATmega1284P]
platform = atmelavr
board = ATmega1284P
framework = arduino
lib_ldf_mode = chain+

```

MCUdude, 2019-08-26:
@ivankravets A new release of MightyCore is now available! Should fix this issue once and for all 🎉

MCUdude, 2019-08-30:
Issue is fixed now. Simply run `pio update`!

closed 2019-08-30 by MCUdude