MCUdude/MightyCore #141: problem with timer0 interrupt at ATmega32

Master811129, 2019-06-05:
hello all.
i have a problem with timer0 interrupt at atmega32 
here is my code:

`#define segPort PORTD
#define shiftPort PORTB
#include <util/delay.h>
#include <avr/interrupt.h>
#include <stdlib.h>
#include <stdint.h>

char display[11] = {0x3f, 0x06, 0x5b, 0x4f, 0x66, 0x6d, 0x7d, 0x07, 0x7f, 0x6f ,0x00};
char num[5] = {1, 2, 3, 4 , 0};
volatile uint8_t i = 0;
uint8_t T = 0;

ISR(TIMER0_OVF_vect) //Timer 0 overflow vector - this run every time timer0 overflows
{
   segPort = display[num[i]];
  _delay_ms(4);
  shiftPort = shiftPort << 1;
  i++;
  if (i > 3)
  {
    i = 0;
    segPort = 0;
    shiftPort = 0x01;
  }
  T++;
}

void setup()
{
  DDRD = 0xff;
  PORTD = 0x00;
  DDRB = 0xff;
  PORTB = 0x01;
  TCCR0=(1<<CS02) | (0<<CS01) | (1<<CS00);
  TCNT0=0xB2;
  TIMSK=(0<<OCIE2) | (0<<TOIE2) | (0<<TICIE1) | (0<<OCIE1A) | (0<<OCIE1B) | (0<<TOIE1) | (1<<TOIE0);
  sei();
  analogReference(INTERNAL);
}

void loop()
{
  if (T >= 250)
  {
    T = 0;
    for (uint8_t del ; del<4;del++)num[del] = 0;
    int analogValue = 0;
    for(uint8_t sampling = 0 ; sampling < 10 ; sampling++)analogValue+=analogRead (A0);
    analogValue = analogValue / 10;
    float voltage = analogValue * 0.002502;
    int temp = int (voltage * 100) + 2;
    itoa(temp, num, 10);
    for (char z = 0 ; z < 4 ; z++)
    {
      if(!isdigit(num[z]))num[z] = 10;
      else num[z] = num[z] - 48;
    }
  }
}`

and it doesn't compile
errors:

wiring.c.o (symbol from plugin): In function `__vector_11':

(.text+0x0): multiple definition of `__vector_11'

C:\Users\Master\AppData\Local\Temp\arduino_build_356334\sketch\segment.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ATmega32.

MCUdude, 2019-06-05:
You will get the same error on every other AVR based Arduino board too. This is because timer0 is used to generate millis and micros.

You can disable all such built-in Arduino functionality by not using void setup() and void loop(), and instead, use int main(void).

``` c
int main(void)
{
  // Setup code here

  while(1)
  {
    // Loop code here
  }
}

```

Master811129, 2019-06-05:
thank you for your help
but does the millis function generate interrupt even when we don't use it?

Master811129, 2019-06-05:
> You will get the same error on every other AVR based Arduino board too. This is because timer0 is used to generate millis and micros.
>
> You can disable all such built-in Arduino functionality by not [...]

thank you for your help man.
but does the millis function generate interrupt even when we don't use it?

MCUdude, 2019-06-05:
> but does the millis function generate interrupt even when we don't use it?

Yes, it does.  Using int main(void) solves this problem

closed 2019-06-05 by MCUdude