MCUdude/MightyCore #146: UART1 on ATmega644A unavailable

JeWe37, 2019-07-02:
Despite the ATmega644A(in contrast to the ATmega644) having a second UART port, Serial1 is not defined, as there is no distinction made between them. For me simply selecting the P versions in the menu tweaking the avrdude.conf so as to have the correct device ID does appear to work. One might just want to list the A with the P and PA rather than the basic 644.

MCUdude, 2019-07-02:
What version of MightyCore are you using? I'm able to compile a sketch with `Serial1` when selecting 644/A in the tools menu

JeWe37, 2019-07-02:
I was on v2.0.1. With the new version it does indeed compile! Well alright then...

MCUdude, 2019-07-02:
For reference, ATmega644 was changed to ATmega644a in commit 4d65c3b.
So, did it actually work for you (on the hardware itself)?

MCUdude, 2019-07-03:
Please re-open if you experience any issues!

closed 2019-07-03 by MCUdude