MCUdude/MightyCore #147: _BV was not declared in this scope while trying to compile Marlin

dc740, 2019-07-07:
The error log is extensive, but it simply complains about _BV being out of scope. This can be reproduced by compiling Marling on branch bugfix-2.0.x (I haven't tried the 1.1 branch) 

```
Arduino: 1.8.9 (Linux), Board: "ATmega1284, Yes (UART0), Standard pinout, 1284P, BOD 2.7V, LTO disabled, External 16 MHz"

In file included from /home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:19:0,
                 from /home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/Arduino.h:290,
                 from sketch/src/lcd/dogm/../../inc/../HAL/HAL_AVR/../shared/Marduino.h:33,
                 from sketch/src/lcd/dogm/../../inc/../HAL/HAL_AVR/HAL.h:27,
                 from sketch/src/lcd/dogm/../../inc/MarlinConfig.h:30,
                 from sketch/src/lcd/dogm/HAL_LCD_class_defines.h:24,
                 from sketch/src/lcd/dogm/ultralcd_DOGM.h:31,
                 from sketch/src/lcd/dogm/lcdprint_u8g.cpp:14:
/home/dc740/arduino-1.8.9/hardware/tools/avr/avr/include/avr/power.h: In function 'void clock_prescale_set(clock_div_t)':
/home/dc740/arduino-1.8.9/hardware/tools/avr/avr/include/avr/power.h:1511:31: error: '_BV' was not declared in this scope
     uint8_t __tmp = _BV(CLKPCE);
                               ^
In file included from /home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:14:0,
                 from /home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/Arduino.h:290,
                 from sketch/src/lcd/dogm/../../inc/../HAL/HAL_AVR/../shared/Marduino.h:33,
                 from sketch/src/lcd/dogm/../../inc/../HAL/HAL_AVR/HAL.h:27,
                 from sketch/src/lcd/dogm/../../inc/MarlinConfig.h:30,
                 from sketch/src/lcd/dogm/HAL_LCD_class_defines.h:24,
                 from sketch/src/lcd/dogm/ultralcd_DOGM.h:31,
                 from sketch/src/lcd/dogm/lcdprint_u8g.cpp:14:
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h: In function 'void enableSleep()':
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:81:5: error: '_BV' was not declared in this scope
     sleep_enable();
     ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h: In function 'void disableSleep()':
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:90:5: error: '_BV' was not declared in this scope
     sleep_disable();
     ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h: In function 'void sleepMode(uint8_t)':
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:102:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_IDLE);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:107:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_ADC);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:112:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_PWR_DOWN);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:117:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_PWR_SAVE);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:122:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_STANDBY);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:127:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_EXT_STANDBY);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h: In function 'void sleep()':
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:148:5: error: '_BV' was not declared in this scope
     sleep_mode();
     ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:148:5: error: '_BV' was not declared in this scope
     sleep_mode();
     ^
In file included from /home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:19:0,
                 from /home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/Arduino.h:290,
                 from sketch/src/lcd/dogm/../../inc/../HAL/HAL_AVR/../shared/Marduino.h:33,
                 from sketch/src/lcd/dogm/../../inc/../HAL/HAL_AVR/HAL.h:27,
                 from sketch/src/lcd/dogm/../../inc/MarlinConfig.h:30,
                 from sketch/src/lcd/dogm/HAL_LCD_class_defines.h:24,
                 from sketch/src/lcd/dogm/ultralcd_DOGM.h:31,
                 from sketch/src/lcd/dogm/ultralcd_DOGM.cpp:42:
/home/dc740/arduino-1.8.9/hardware/tools/avr/avr/include/avr/power.h: In function 'void clock_prescale_set(clock_div_t)':
/home/dc740/arduino-1.8.9/hardware/tools/avr/avr/include/avr/power.h:1511:31: error: '_BV' was not declared in this scope
     uint8_t __tmp = _BV(CLKPCE);
                               ^
In file included from /home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:14:0,
                 from /home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/Arduino.h:290,
                 from sketch/src/lcd/dogm/../../inc/../HAL/HAL_AVR/../shared/Marduino.h:33,
                 from sketch/src/lcd/dogm/../../inc/../HAL/HAL_AVR/HAL.h:27,
                 from sketch/src/lcd/dogm/../../inc/MarlinConfig.h:30,
                 from sketch/src/lcd/dogm/HAL_LCD_class_defines.h:24,
                 from sketch/src/lcd/dogm/ultralcd_DOGM.h:31,
                 from sketch/src/lcd/dogm/ultralcd_DOGM.cpp:42:
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h: In function 'void enableSleep()':
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:81:5: error: '_BV' was not declared in this scope
     sleep_enable();
     ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h: In function 'void disableSleep()':
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:90:5: error: '_BV' was not declared in this scope
     sleep_disable();
     ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h: In function 'void sleepMode(uint8_t)':
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:102:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_IDLE);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:107:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_ADC);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:112:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_PWR_DOWN);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:117:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_PWR_SAVE);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:122:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_STANDBY);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:127:9: error: '_BV' was not declared in this scope
         set_sleep_mode(SLEEP_MODE_EXT_STANDBY);
         ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h: In function 'void sleep()':
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:148:5: error: '_BV' was not declared in this scope
     sleep_mode();
     ^
/home/dc740/.arduino15/packages/MightyCore/hardware/avr/2.0.2/cores/MCUdude_corefiles/wiring_extras.h:148:5: error: '_BV' was not declared in this scope
     sleep_mode();
     ^
exit status 1
Error compiling for board ATmega1284.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

MCUdude, 2019-07-07:
I have no experience with the Marlin FW. What happens if you add `#include <avr/sfr_defs.h>` to the top of the INO file?

dc740, 2019-07-07:
I get exactly the same error

dc740, 2019-07-07:
OK, I got around it by manually defining _BV in the power.h file. So, basically, adding `#include <avr/sfr_defs.h>` inside of `~/arduino-1.8.9/hardware/tools/avr/avr/include/avr/sfr_defs.h` seems to be a good workaround. Adding it at the top of the .ino file made no difference

msalajka, 2019-07-17:
i've tried it (Marlin bugfix-2.0.x) with Ender 3 (board v1.1.4).

good place to put fix is:
 ..\packages\MightyCore\hardware\avr\2.0.2\cores\MCUdude_corefiles\Arduino.h

#ifndef _BV
#define _BV(bit) (1 << (bit))
#endif

MightyCore/ATmega1284 setup:
- Clock External 16 MHz
- BOD disabled
- LTO disabled
- Sanguino pinout
- 1284P variant
- Bootloader UART0

bootloader burned via USBasp, Marlin firmware uploaded via usb.
Sketch uses 129972 bytes (99%) of program storage space. 
With LTO enabled, sketch uses 122708 bytes (94%) of program storage space.

It works ok. ;)

dc740, 2019-07-18:
Same here. I updated my comment to reflect the fact that I was doing it wrong. I enabled BOD to 2.7 and it works OK too

MCUdude, 2020-05-15:
I'm curious. I've recently upgraded my old DIW Prusa i3 from 2014 with a brand new 32-bit board. I've also used PlatformIO to build the latest Marlin. But why are you using MightyCore for building for the ATmega1284P? I mean, that's completely fine, but I thought Marlin favored the Sanguino core? Is it to save flash?

SpenceKonde, 2020-05-24:
_BV should be defined by the core, it is widely used in existing code.

MCUdude, 2020-05-24:
What's weird is that _BV works great in Arduino IDE and in PlatformIO projects. There's something with the Marlin build that's causing this. However, I haven't really looked into it.

> I would characterize this a bug in the core.

In the core like in *this* core?

SpenceKonde, 2020-05-24:
Wait, I assumed the Marlin FW was being compiled within the Arduino IDE
context - is it not?

MCUdude, 2020-05-24:
For 8-bit AVR boards you can use both. For 32-bit you can only use PlatformIO. However, I don't see any reason not to use PlatformIO for a project as large as this. PlatformIO/VSCode is much better at handling multiple files.

SpenceKonde, 2020-05-24:
You can say that again (having just thrown two weeks of precious
quarantimes into the rathole called jtag2updi, when I *wanted* to be
working on 'da core y'all been waiting for - compiling with Arduino IDE
since I now know that shitshow like the back of my hand* )

I have such amazing plans for Delta Alpha Charlie oscar romeo echo - that
shit is gonna be fucking BALLER! That memory section shit, aw man - that
was the missing piece of the puzzle! I'm gonna use the "fast IO" trick to
make a fast-pinmode (I'm gonna bring that to megaTinyCore, and you can ofc
steal it too - but I can make even normal pinmode calls faster on these new
parts because of that pinctrl setting register (i find it HILARIOUS that
they put warnings about how confusing the set and clear registers were,
considering the bloody important stuff they neglected to put such a warning
by and which is like, mad counterintuitive - like, FFS they didn't warn
people about the wacky behavior of the third byte in the UPDI address
pointer, they don't tell people that bits 1 and 7 in the ASI status
register mean something, or even how to check the silicon rev**  - yet they
warn people about the set and clear registers? What?!)

Maybe that's a little pejorative - it was a rabbit hole for the first 3-4
days, but since then, it's been a straight up rathole, and I am so sick of
it (that's why I haven't been around - though the new jtag2updi will be way
better than the old one. It doesn't glitch out and hang like it used to,
and it also doesn't run first dozen milliseconds or so of the sketch
several times during the programming process either, and it programs
DA-series parts, and it has a proper debug channel (via SPI if the thing
it's running on doesn't have a second USART - could *not* have made it work
without that! I mean, I made rudimentary debug work on tinyAVR 0-series and
1-series parts via pinswapping, but that... was not fit for purpose in
real-world conditions...). It also disables response signature during burst
writes to improve throughput.

* Actually, I'm pretty sure I know the Arduino IDE better than the back of
my hand right now, because the backs of my hands are doing things I don't
understand - breaking out in hives for no apparent reason - hives which
then pop up on legs and feet after the hands, and when they do my feet, my
feet swell up like balloons, itch and hurt like a mofo, and I can barely
walk (my situation right now - but what am I supposed to do, burn a $20
health-insurance copay for some useless quack to look at it over zoom?
Yeah, when pigs fly...)

** A particularly egregious omission, what with the silicon errata for the
current rev of you know what parts being enough to bring a grown man to
tears!

c---, 2022-05-19:
The problem is caused by dependency cycles in MightyCore(?). I don't know where all of them are but at least between `wiring_extras.h` and `Arduino.h` which both include each other.

The solution was simply to remove the `#include "wiring_extras.h"` from Arduino.h. I don't know all the use cases for MightyCore so I don't know if this is a fix for anything other than my version of Marlin. Someone should probably spend some time "flattening" the dependency graph in the code base as this could both prevent errors and possibly speed up compilation.

If you enable verbose compilation in Arduino you can just copy/paste any (failing) gcc command and add the `-E` option to see what's going on in the preprocessor. This output is difficult to parse though. I don't know if there are maybe tools or options that can show everything in a more organized way.

closed 2026-02-10 by MCUdude