MCUdude/MightyCore #150: can not burn bootloader

arifainchtein, 2019-07-24:
Hell,
I am wanting to use a 1284P and burn the bootloader on to it using an Arduino uno as ISP.  I have followed your instructions and install the MightyCore. Everything seems ok, as you can see from the first image
[image]

However, when i try t burn the bootloader, I get an error.  The weird thing is that avrdude is giving the erro that it can not find the USBtiny device.

[image]

Do you have any suggestions as to what could be wrng with my setup?

Thanks

MCUdude, 2019-07-24:
**ArduinoISP** and **Arduino as ISP** is two different type of programmers.

You need to select Arduino as ISP as your programmer.

MCUdude, 2019-08-02:
Did this solve the issue?

arifainchtein, 2019-08-08:
sorry, i missed it, yes this resolve the issue. thanks fr the help

closed 2019-08-08 by arifainchtein