MCUdude/MightyCore #154: Interrupts not working on Atmega644pa

cpsinghopl, 2019-08-06:
Hello,

Thanks for providing the library. 
Flashed Atmega644PA using ICSP on Arduino UNO (code written & working on UNO), however interrupts are not working. Rest all other functionality is working.
Is there anything extra which needs to be done?

FYI. using Bobuino pinout, LTO disabled, External 16MHZ clock.

Regards

MCUdude, 2019-08-06:
What package is your 644PA, and what physical pins are you using for interrupts?

PiotrC1, 2019-08-10:
It's interresting as I use 644PA and interrupts are working in my case.

MCUdude, 2019-08-26:
I just tested the interrupt pins on a 644PA with both Standard and Bobuino pinout. Everything works just fine. Pretty confident it's not related to MightyCore.

closed 2019-08-26 by MCUdude