MCUdude/MightyCore #155: <Wire.h> may not linked well in version 2.0.2,

randika2000e, 2019-08-07:
<Wire.h> may not linked well in version 2.0.2, 
'BUFFER_LENGTH' was not declared in this scope
unable to retrieve BUFFER_LENGTH size from the <Wire.h>
but 2.0.1 version working well.

MCUdude, 2019-08-07:
BUFFER_LENGTH has purposely been renamed to [TWI_BUFFER_SIZE](https://github.com/MCUdude/MightyCore/blob/152885b1b8b10a3695f0241c6af196482e7e53f4/avr/libraries/Wire/src/utility/twi.h#L32) in v2.0.2.

MCUdude, 2019-08-07:
I chose to add it again to keep backwards compatibility for code that might depend on BUFFER_LENGTH.

closed 2019-08-07 by MCUdude

randika2000e, 2019-08-07:
The "BUFFER_LENGTH" should be include into the Wire.h file, not into the twi.h. please see the 2.0.1 Wire.h file

// Only include Wire1 if two i2c channels are available
#if defined(TWI1_vect)
  #include <Wire1.h>
#endif

#define BUFFER_LENGTH 32

MCUdude, 2019-08-07:
> // Only include Wire1 if two i2c channels are available
#if defined(TWI1_vect)
#include <Wire1.h>
#endif

This code along with .a linkage did not work at all in PlatformIO, that's why it's reverted in v2.0.2. If you want to use Wire1, you'll manually have to include Wire1.h

> The "BUFFER_LENGTH" should be included in the Wire.h file, not into the twi.h. please see the 2.0.1 Wire.h file

So you're not able to compile your code because the twi buffer length is included in twi.h, not Wire.h?

commit c3c55b7f (2019-08-07): Add BUFFER_LENGTH to Wire library
Keeps backwards compatibility and closes #155