MCUdude/MightyCore #157: Atmega 644p wrong LED for Bobuino pinout

PiotrC1, 2019-08-10:
Hi,
In Bobuino pinout LED is attached to PB7 but it's disregarded by bootloader downloaded with programmer. In bootloader LED is hardcoded as PB0. It seems it has to be separate bootloader for Bobuino pinout (I think not only for Atmega644 but for other chips as well).

MCUdude, 2019-08-16:
I'm fully aware of this, but I don't want to provide additional bootloaders since the existing bootloader names don't have the physical LED pin in their file names.

Let's call it a known flaw.

mrguen, 2019-11-02:
Hello,

I have the same problem. I could not compile a bootloader on Windows 7. Some files are not found when trying to run make even though I added the PATH in windows. Does anyone compiled for the LED on PB7? (@PiotrC1)

mrguen, 2019-11-02:
Ok, here is what you need to do to compile under Windows. For example we compile for Atmega1284p / 16 MHz 115000 bauds, led = B7, with EEPROM and copy_flash:

* Save the old bootloaders
* Clean the directory
C:\Users\TG\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.3\bootloaders\optiboot_flash\bootloaders\atmega1284p\16000000L 
* Open a cmd window
* cd C:\Users\TG\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.3\bootloaders\optiboot_flash
* make atmega1284p AVR_FREQ=16000000L BAUD_RATE=115200 LED=B7 LED_START_FLASHES=2 UART=0 BIGBOOT=1

PiotrC1, 2019-11-02:
Hmm, did you really compile that way? I had to modify Makefile line 59 `TOOLROOT = /some/windows/path`

mrguen, 2019-11-03:
@PiotrC1: indeed. I just updated the method.

mrguen, 2019-11-03:
@MCUdude 

Hi, I just modified

* boards.txt
* Makefile

So the bootloader would have the LED number included in its name and the right BUILTIN_LED will be selected when selecting the specific Pinout. 

I also addded the Narrow pinout that I am working on.

But did not generate the 9460 bootloaders...

Would you like to update MightyCore with theses?

[mighty.zip](https://github.com/MCUdude/MightyCore/files/3801392/mighty.zip)

MCUdude, 2019-11-03:
@mrguen Thanks, but I can't just edit the makefile without updating all my other Arduino cores as well.

I'm kinda busy at the moment, but I'll update the makefile and the makeall script in the [Optiboot flash](https://github.com/MCUdude/optiboot_flash) repo. Then I can set up the script to build extra "Bobuino compatible" bootloader files for ATmega1284/P, ATmega644A/P, ATmega324A/P/PA/PB, ATmega164A/P, ATmega32, ATmega16 and ATmega8535.

After this is done I can update the Optiboot flash subtree in MightyCore, MiniCore, MegaCore, and MajorCore afterward. Then the appropriate changes has to be done in every boards.txt file.

I will use your implementation where the bootloader LED is chosen based on the pinout. However, I will not add another pinout, since I think we have enough pinouts to chose from already.

mrguen, 2019-11-03:
@MCUdude ok, let me know whenever you can do it. I might help you changing the boards.txt files.

MCUdude, 2019-11-03:
I'll first have to discuss this with the PlatformIO developers. I'm working on a project where the correct fuse bits and bootloader files can be calculated and loaded automatically (based on the user settings in the platformio.ini file), but a change in the bootloader name (adding the LED pin) will break all this. I'm sure we can solve this quite easily, but I'd like to hear what they have to say about this first.

MCUdude, 2019-11-04:
I've now pushed a fix. Please try to install the latest version of MightyCore. A boards manager release will be available when this is changed on all Arduino cores

closed 2019-11-10 by MCUdude

PiotrC1, 2019-11-11:
Hmm - the version is still 2.0.3 (in `https://MCUdude.github.io/MightyCore/MightyCore-2.0.3.tar.bz2` so my Arduino IDE doesn't update.

MCUdude, 2019-11-11:
I haven't done a release yet. For now you will have to do a manual install.

DirtyEngineer, 2020-02-28:
@mrguen Thank you for the `TOOLROOT = 'C:/Program Files (x86)/Arduino/hardware/tools/avr'` fix for Windows bootloader builds. I couldn't get the bootloaders in the MegaCore to build and this fixed my problem.

mrguen, 2020-02-28:
@DirtyEngineer your welcome. @MCUdude It's a problem not to have the wew release published. It's confusing.