MCUdude/MightyCore #159: Problems with booloader compilation - Linux

PiotrC1, 2019-08-26:
In order to solve or better to detour issue #157 I tried to compile myself bootloaders. In Linux (with installed IDE 1.8.9 file bootloaders/optiboot_flash/Makefile has incorrect access rights - should be executable and is not.

Also escape codes should be echoed using echo -e (instead of pure echo) and echoed string should be single quoted instead of being double quoted.

MCUdude, 2019-08-27:
You can make the makeall script executable by running `chmod +x ./makeall`.

> Also escape codes should be echoed using echo -e (instead of pure echo) and echoed string should be single quoted instead of being double quoted.

I'm no bash expert at all. I made the makeall script so it would be easier to compile lots of files for lots of targets. Why would this matter for the functionality of the script?

MCUdude, 2019-08-27:
Reading closer, it seems like you're not using the makeall script at all. You still can't execute the makefile directly. Run instead a modified version of this command:

`make atmega8 AVR_FREQ=16000000L BAUD_RATE=115200 LED=B5 LED_START_FLASHES=2 UART=0`

PiotrC1, 2019-08-27:
Well, after changing `./Makefile` (not `./makeall` as this one has correct access rights) to be executable you can compile everything with one command - `make`

Escape codes do not influence functionality, but it looks ugly without escape codes working correctly (some text is not underlined).

But still you have to modify line
> TOOLROOT = ArduinoInstallPath/hardware/tools/avr 
placing correct path to Aruino files.

MCUdude, 2019-08-27:
> Well, after changing ./Makefile (not ./makeall as this one has correct access rights) to be executable you can compile everything with one command - make

I'm not sure I follow. Did you run `chmod +x Makefile` and then `./Makefile`? That wouldn't work.
FYI if you run `make` on a Mac or Linux computer the makeall script is automatically executed.

> Escape codes do not influence functionality, but it looks ugly without escape codes working correctly (some text is not underlined).

Well, it looks perfectly fine in my terminal? What text do you mean should be underlined that isn't? Could you post a screenshot?

> But still you have to modify line
TOOLROOT = ArduinoInstallPath/hardware/tools/avr
placing correct path to Aruino files.

Of course! It's impossible to predict where the IDE install is placed on a linux computer. It's different on a Mac for instance. It could need a proper description in the [README](https://github.com/MCUdude/optiboot_flash) though.

PiotrC1, 2019-08-27:
> I'm not sure I follow. Did you run chmod +x Makefile and then ./Makefile? That wouldn't work.
FYI if you run make on a Mac or Linux computer the makeall script is automatically executed.

Misunderstanding. I use Ubuntu LTS 18.04. I have installed newest (1.8.9) Arduino IDE, as well as MightyCore 2.0.2.  If I issue comand  `make` I got some error. If I change `Makefile` to be executable (`chmod +x Makefile`) and then issue `make` everything runs smoothly.

> Well, it looks perfectly fine in my terminal? What text do you mean should be underlined that isn't? Could you post a screenshot?

OK. Give me some time, as I'm away from my Ubuntu laptop.

BTW. I managed to compile single bootloader under Windows environement.

jandrassy, 2019-08-27:
sorry, just of curiosity, why do you compile the bootloader?

MCUdude, 2019-08-27:
> Misunderstanding. I use Ubuntu LTS 18.04. I have installed newest (1.8.9) Arduino IDE, as well as MightyCore 2.0.2. If I issue command make I got some error. If I change Makefile to be executable (chmod +x Makefile) and then issue make everything runs smoothly.

Thanks for the clarification! I personally have never needed to make the makefile executable. I'm not sure why yours weren't though.

> sorry, just of curiosity, why do you compile the bootloader?

@jandrassy it's because he wants the LED pin to be something else than PB0.

PiotrC1, 2019-08-27:
Good question. I have made reversed osmosis controller with a board similar to Arduino Nano Pro but with Atmega644p (Nano was too small). Pinout is Bobuino (i.e. LED is connected to B7). To B0 (which is used as LED output for other pinouts - Standard and Sanguino) I have connected inlet valve. Every time I download a new program this valve open twice what is very unwanted.

MCUdude, 2019-11-10:
@PiotrC1 I guess I can close this issue?

PiotrC1, 2019-11-11:
Hi @Hans,

                The version is still 2.0.3 so my Arduino IDE doesn’t update!

 

Brg,

Piotr

MCUdude, 2019-11-11:
> The version is still 2.0.3 so my Arduino IDE doesn’t update!

There isn't anything wrong with the code in release v2.0.3. You were able to compile a bootloader after making the makefile executable. What more can I do?

MCUdude, 2020-03-15:
MightyCore v2.0.4 is now available and should have all this sorted out.

closed 2020-03-15 by MCUdude