MCUdude/MightyCore #163: Bootloader compatibility with USB/UART converters

mrguen, 2019-09-18:
Hello,

Thanks for this core. It seems very interesting. I am having troubles making this Atmega644 barebone:

Atmega644PA-MU
Resonator Murata 16 MHz
CH340G linked to the MCU through 1 Kohm on Rx, Tx + DTR through 100 nf to reset.
Reset pulled up (10kOhm) to 5V.

I could assemble one board. Everything running fine, but then I assembled two others. None is working as expected. The 3 MCUs were bought together on aliexpress. 

* Firstly:  2 factory fresh MCUs would not communicate at all. Finally I could start them with a 1 MHz external clock. So apparently they were "factory bricked" and everything went fine after loading the fuse bytes. Or there is something with the loading of the bootloader on factrory fresh board.

* Secondly: After starting the device and loading the botloader using USBAsp, with this configuration:
MightyCore 2.0.3 
Clock: external 16 MHz
Variant: 644PA
Bootloader: yes UART0
Leds between 5V-330Ohms and Rx and Tx to visualize the signals

I could not make it work properly with an FTDI USB/UART adapter and not at all with a CH340G:

* FTDI: connected with Rx, TX and DTR through a 100 nf to reset. **Works fine once** and then is out of sync. Like if loading the sketch over the USB erased the bootloader. 

* CH340: connected with Rx, TX and DTR through a 100 nf to reset.: out of sync. Never worked.

Both adapters are well enumerated by Windows.

I have checked everything:  
* oscillators (on MCU and CH340) are running. The MCU seems to be working at the expected frequency 
* fuse bytes, 
* good connections of D+, D-,  Rx, Tx, Reset etc...
* MCU can run blink when loaded through USBAsp
* Voltages are good: 5.1V and stable
* changed the CH340G

So there remains some possibilities:

a) The bootloader does not manage well USB/UART devices. In some cases it works, others not... maybe a too tight timing or something... 
- Argument in favor of this hypothesis: it is possible to use the FTDI adapter once and then it stops working. So apparently the chip is ok / can load a program though an USB/UART adapter.

b) Those chips are not working as expected. But 
- they run normally in 1 MHz internal or 16 MHz external, 
- they can load and run blink on a programmer (USBAsp)

I really don't get what is happening. One board is working properly so maybe this configuration is at the physical limits and some are working others not / the chips are altered but still working. I have seen many posts on the web about people having the impression they bricked factory fresh devices... is there a known bad production batch?
- Argument in favor of this hypothesis: there was something with these chips when I first tried to uplod the bootloader. They would not communicate at all.

c) The murate resonator is not precise enough. But why would hypothesis a) be working once. Also this is a usual clock circuit (on the Nano etc)

Overall, since the behaviour is not constant I suspect there is something with the mightycore / the bootloader, that might have a bug on some timing parameters set to a wrong value on certain conditions.

MCUdude, 2019-09-18:
> FTDI: connected with Rx, TX and DTR through a 100 nf to reset. Works fine once and then is out of sync. Like if loading the sketch over the USB erased the bootloader.

This tells me that you have an issue with the auto-reset circuity. When the bootloader is continuing to run the bootloader until a program is loaded. If you power-cycle the chip, it exits the bootloader within a second of no data was received.

Can you connect an LED to digital pin 0 (PB0) and manually reset the board? The LED should flash twice. This also indicates that a bootloader is present.

If you have an oscilloscope you can probe the RST line on the microcontroller to see if the DTR line is able to pull the RST line low. Are you 100% sure you're using a 100nF (often marked 104) and not some other, lower value?

MCUdude, 2019-09-18:
I've sold more than MightyCore development boards where a CH340C/G chip is used for serial communication. None of them have experienced what you're describing. Which again strengthens my theory.

mrguen, 2019-09-18:
PB0 is flashing twice every second.

MCUdude, 2019-09-18:
> PB0 is flashing twice every second.

OK. That means that the bootloader is present. Check your auto-reset circuity and make the LEDs with their "low value" 330R resistors doesn't interfere with the RX/TX line. If the USB to serial adapter is also running at 5V you can and should remove the 1k series resistors just to rule out any additional errors.

mrguen, 2019-09-18:
Indeed, i tested this with the FTDI adapter:
- DTR- 100 nf - reset:  * works once. * works when manually resetting 
- DTR - 1 uF - reset: works

So maybe my circuit is too low impedance as you said. Also the timing is maybe not good.

MCUdude, 2019-09-18:
I'm not worried abo timing and the accuracy of the Murata resonator. The company I work for literally have thousands of products where the micricontrollers are running on Murata CSTNE resonators, and they are communicating over RS485. USART is pretty forgiving, so less than 3% error will work.

I'm pretty sure it's about the impedance. If it works with a 1uF capacitor it means that it's resistance/impedance to 5V is too low. Are you 100% sure you use a 10k and not a 1k pullup resistor?

MCUdude, 2019-09-18:
But it's not really important to use a 100nF capacitor. What's important is that it should be able to pull the reset line low, and a 100nF cap is suited for this. A 1uF is also OK, but it may be a good idea to add a diode (1n4148, BAT54 etc.) to clamp the voltage to 5V. When the DTR line goes from low to high again you may get voltage doubling, resulting in 10V on the RST pin for a few milliseconds. This may cause the microcontroller to enter high voltage programming mode, and you don't want that.

mrguen, 2019-09-18:
Ok, I tried with the CH340
- 1 uF between DTR and reset : not working
- manual reset: not working.

Then I monitored the signals using salae Logic. Either the signals from the CH340 are too week or the lines are pulled down because I can see the RX signal on the CH340 side of the series resistor but not on the MCU side.

 there is no activity at all from the CH340G even though it is recognized by windows it does not seem to be working.

MCUdude, 2019-09-18:
The logic analyzer won't show you the analog values, so at this point, it's pretty useless for what you're trying to solve. If you have access to one, use an oscilloscope.

Why have the resistors there?

mrguen, 2019-09-18:
This is how it is cabled on the Nano clones (with 1k series). Do you mean I should be looking for ringing or something like that?

MCUdude, 2019-09-18:
No, ringing won't be an issue. You need to monitor the lines to see if they really are pulled low or not. On an oscilloscope, you can easily see if the USB to serial adapter is "fighting" a low impedance pin that's forced high. If you don't see the same signal on both sides of the resistor that's probably what's happening.

mrguen, 2019-09-18:
Why couldn't I see it with a logic analyzer? I can see perfectly when it is working properly 
the Rx and Tx signals.

MCUdude, 2019-09-18:
Not if the levels of the actual signal cross the threshold level of the logic analyzer. That is the difference between it and a scope. But this isn't really a problem with mightycore, but rather a hardware issue. I suggest asking on the Arduino forums instead, where more people can help

mrguen, 2019-09-18:
I realized when reading this https://forum.arduino.cc/index.php?topic=537131.0 that the serial monitor is working !

But the solution provided of deactivating the fifo buffer does not work for me...

mrguen, 2019-09-18:
I tried to modify the high fuse because the boot flash section seemed to small 0xD6 is for 512 words while the maximum size is specified at 64512 bytes. So I changed for the defualt value 0XD0 = 4096 words. But this does not solve the issue.

mrguen, 2019-09-19:
Ok, there was indeed a problem in my circuit with the led implementation. I don't think the Rx and Tx led were the problem but a led on SCL... I am not totally certain but I will come back to a regular schematics. Thanks for the support.

closed 2019-09-19 by mrguen