MCUdude/MightyCore #165: MightyCore 1284 - Project will not compile if PWM.h is included

JacksonAudio, 2019-09-26:
The project will not compile if PWM.h is included. I get this error if I run any version of MightyCore or MiniCore.

129:4: error: 'SetPinFrequency' was not declared in this scope

Please help. I need the speed of either the 1284 or 328 running at 20MHz and I can't find a way to get there. If I compile for the Uno or the MEGA 2560 it works great. 

I'm more than happy to donate to the cause if we can find a solution to this problem.

Brad Jackson
brad@jacksonaudio.net

MCUdude, 2019-09-26:
And what library is PWM.h?

I'm pretty sure the library you're using isn't designed to work on an ATmega1284P. However, that's not MightyCore's fault.

JacksonAudio, 2019-09-26:
Hello Hans!

Thanks so much for writing back. 

Here is where you can find it: https://code.google.com/archive/p/arduino-pwm-frequency-library/downloads <https://code.google.com/archive/p/arduino-pwm-frequency-library/downloads>

I wasn’t trying to imply that it was MightyCore’s fault. I just can’t get MC and PWM.h to play nice together and would really appreciate some help if you’re able.

Again I’m happy to pay if you can get the 328 and 1284 to talk to PWM.h

If you download the library and run their sample, with either the 1284 or 328 it should break.

Thanks again!

		BRAD JACKSON 
President / CEO

JACKSON AUDIO / JACKSON AMPWORKS

email: brad@jacksonaudio.net <mailto:brad@jacksonaudio.net> 
phone: 1-559-JACKSON (522-5766) 
web:  http://jackson.audio <http://jackson.audio/> 

110 Sports Parkway, Suite C
KELLER, TX 76248

 <https://facebook.com/jacksonaudio>	 <https://youtube.com/jacksonaudio>	 <https://instagram.com/jacksonaudio>	 <https://twitter.com/jackson_audio>

MCUdude, 2019-09-26:
I just had a look at the library, and it's _very spesificly_ written for ATmega48/88/168/328 and ATmega640/1280/2560. In fact, the library would pretty much have to be rewritten in order to support the ATmega1284P. It certainly can be done, but I don't have time to do it.

But what are you trying to accomplish? You don't need to use a library if you want to change the PWM frequency.

JacksonAudio, 2019-09-26:
My whole project is based around the idea of changing the PWM frequency.
I’m adjusting the clock frequency going to a timing chip.

Thanks so much for the information and your time.
It is greatly appreciated.

All the best!

		BRAD JACKSON 
President / CEO

JACKSON AUDIO / JACKSON AMPWORKS

email: brad@jacksonaudio.net <mailto:brad@jacksonaudio.net> 
phone: 1-559-JACKSON (522-5766) 
web:  http://jackson.audio <http://jackson.audio/> 

110 Sports Parkway, Suite C
KELLER, TX 76248

 <https://facebook.com/jacksonaudio>	 <https://youtube.com/jacksonaudio>	 <https://instagram.com/jacksonaudio>	 <https://twitter.com/jackson_audio>

MCUdude, 2019-09-26:
I see. What PWM pin on the 1284P are you planning to use? The maximum speed on some pins (when running at 16 MHz) is ~64 kHz, while others are ~32 kHz.

JacksonAudio, 2019-09-26:
I would pick the two that allow the fastest operation.
I think 30kHz would be the fastest I would need honestly.

Also for clarification, I’m not using the P version of the chip and I’m also running it at 20MHz.

Thanks!

		BRAD JACKSON 
President / CEO

JACKSON AUDIO / JACKSON AMPWORKS

email: brad@jacksonaudio.net <mailto:brad@jacksonaudio.net> 
phone: 1-559-JACKSON (522-5766) 
web:  http://jackson.audio <http://jackson.audio/> 

110 Sports Parkway, Suite C
KELLER, TX 76248

 <https://facebook.com/jacksonaudio>	 <https://youtube.com/jacksonaudio>	 <https://instagram.com/jacksonaudio>	 <https://twitter.com/jackson_audio>

MCUdude, 2019-09-26:
Do some research on how you can modify the TCCRxB register, where x is the timer number (0..3). If you don't want to mess up timing functions like millis(), micros() and delay, stay clear of timer 0 and TCCR0B.

You can look it up in the datasheet (search for TCCRnB). Basically, you want no prescaling.

JacksonAudio, 2019-09-26:
Thanks for the information! :)

		BRAD JACKSON 
President / CEO

JACKSON AUDIO / JACKSON AMPWORKS

email: brad@jacksonaudio.net <mailto:brad@jacksonaudio.net> 
phone: 1-559-JACKSON (522-5766) 
web:  http://jackson.audio <http://jackson.audio/> 

110 Sports Parkway, Suite C
KELLER, TX 76248

 <https://facebook.com/jacksonaudio>	 <https://youtube.com/jacksonaudio>	 <https://instagram.com/jacksonaudio>	 <https://twitter.com/jackson_audio>

JacksonAudio, 2019-09-26:
Hans I just had a thought. 

I had this same issue with the MiniCore boards when I tried to use the 328 you have listed there. (Both 16 and 20MHz) Would using the 328 be easier with this library? Frankly I may be able to use the 328 and run it at 20MHz to do what I need to do.

Thanks!

		BRAD JACKSON 
President / CEO

JACKSON AUDIO / JACKSON AMPWORKS

email: brad@jacksonaudio.net <mailto:brad@jacksonaudio.net> 
phone: 1-559-JACKSON (522-5766) 
web:  http://jackson.audio <http://jackson.audio/> 

110 Sports Parkway, Suite C
KELLER, TX 76248

 <https://facebook.com/jacksonaudio>	 <https://youtube.com/jacksonaudio>	 <https://instagram.com/jacksonaudio>	 <https://twitter.com/jackson_audio>

MCUdude, 2019-09-26:
> I had this same issue with the MiniCore boards when I tried to use the 328 you have listed there. (Both 16 and 20MHz) Would using the 328 be easier with this library? Frankly I may be able to use the 328 and run it at 20MHz to do what I need to do.

The ATmega328P is at least supported by the library, so it should be much easier. If you get an error message we should be possible to figure the root cause of the issue.

JacksonAudio, 2019-09-26:
I just got the MiniCore 328 to work at 20MHz! So that’s HUGE progress!

At least now I can run that at 20MHz!

Here are the error messages of the 1284:
Arduino: 1.8.10 (Mac OS X), Board: "ATmega1284, Yes (UART0), Standard pinout, 1284, BOD 2.7V, LTO disabled, External 20 MHz"

/Applications/Arduino 3.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino 3.app/Contents/Java/hardware -hardware /Users/BradJackson/Library/Arduino15/packages -hardware /Users/BradJackson/Google Drive/Jackson Ampworks/Arduino/hardware -tools /Applications/Arduino 3.app/Contents/Java/tools-builder -tools /Applications/Arduino 3.app/Contents/Java/hardware/tools/avr -tools /Users/BradJackson/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 3.app/Contents/Java/libraries -libraries /Users/BradJackson/Google Drive/Jackson Ampworks/Arduino/libraries -fqbn=MightyCore:avr:1284:bootloader=uart0,pinout=standard,variant=modelNonP,BOD=2v7,LTO=Os,clock=20MHz_external -ide-version=10810 -build-path /var/folders/nr/4g_3dt4s3hjfvsy_bgj540c40000gp/T/arduino_build_432006 -warnings=none -build-cache /var/folders/nr/4g_3dt4s3hjfvsy_bgj540c40000gp/T/arduino_cache_51851 -prefs=build.warn_data_percentage=75 -verbose /Users/BradJackson/Google Drive/Jackson Ampworks/Arduino/ECHOBASE_freq_test/ECHOBASE_freq_test/ECHOBASE_freq_test.ino
/Applications/Arduino 3.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino 3.app/Contents/Java/hardware -hardware /Users/BradJackson/Library/Arduino15/packages -hardware /Users/BradJackson/Google Drive/Jackson Ampworks/Arduino/hardware -tools /Applications/Arduino 3.app/Contents/Java/tools-builder -tools /Applications/Arduino 3.app/Contents/Java/hardware/tools/avr -tools /Users/BradJackson/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 3.app/Contents/Java/libraries -libraries /Users/BradJackson/Google Drive/Jackson Ampworks/Arduino/libraries -fqbn=MightyCore:avr:1284:bootloader=uart0,pinout=standard,variant=modelNonP,BOD=2v7,LTO=Os,clock=20MHz_external -ide-version=10810 -build-path /var/folders/nr/4g_3dt4s3hjfvsy_bgj540c40000gp/T/arduino_build_432006 -warnings=none -build-cache /var/folders/nr/4g_3dt4s3hjfvsy_bgj540c40000gp/T/arduino_cache_51851 -prefs=build.warn_data_percentage=75 -verbose /Users/BradJackson/Google Drive/Jackson Ampworks/Arduino/ECHOBASE_freq_test/ECHOBASE_freq_test/ECHOBASE_freq_test.ino
Using board '1284' from platform in folder: /Users/BradJackson/Library/Arduino15/packages/MightyCore/hardware/avr/2.0.3
Using core 'MCUdude_corefiles' from platform in folder: /Users/BradJackson/Library/Arduino15/packages/MightyCore/hardware/avr/2.0.3
Detecting libraries used...
"/Applications/Arduino 3.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega1284 -DF_CPU=20000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -I/Users/BradJackson/Library/Arduino15/packages/MightyCore/hardware/avr/2.0.3/cores/MCUdude_corefiles -I/Users/BradJackson/Library/Arduino15/packages/MightyCore/hardware/avr/2.0.3/variants/standard /var/folders/nr/4g_3dt4s3hjfvsy_bgj540c40000gp/T/arduino_build_432006/sketch/ECHOBASE_freq_test.ino.cpp -o /dev/null
Alternatives for PWM.h: [PWM-master@5.0]
ResolveLibrary(PWM.h)
  -> candidates: [PWM-master@5.0]
"/Applications/Arduino 3.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega1284 -DF_CPU=20000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -I/Users/BradJackson/Library/Arduino15/packages/MightyCore/hardware/avr/2.0.3/cores/MCUdude_corefiles -I/Users/BradJackson/Library/Arduino15/packages/MightyCore/hardware/avr/2.0.3/variants/standard "-I/Users/BradJackson/Google Drive/Jackson Ampworks/Arduino/libraries/PWM-master" /var/folders/nr/4g_3dt4s3hjfvsy_bgj540c40000gp/T/arduino_build_432006/sketch/ECHOBASE_freq_test.ino.cpp -o /dev/null
Alternatives for ArduinoTapTempo.h: [ArduinoTapTempo-master@1.0]
ResolveLibrary(ArduinoTapTempo.h)
  -> candidates: [ArduinoTapTempo-master@1.0]
"/Applications/Arduino 3.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega1284 -DF_CPU=20000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -I/Users/BradJackson/Library/Arduino15/packages/MightyCore/hardware/avr/2.0.3/cores/MCUdude_corefiles -I/Users/BradJackson/Library/Arduino15/packages/MightyCore/hardware/avr/2.0.3/variants/standard "-I/Users/BradJackson/Google Drive/Jackson Ampworks/Arduino/libraries/PWM-master" "-I/Users/BradJackson/Google Drive/Jackson Ampworks/Arduino/libraries/ArduinoTapTempo-master/src" /var/folders/nr/4g_3dt4s3hjfvsy_bgj540c40000gp/T/arduino_build_432006/sketch/ECHOBASE_freq_test.ino.cpp -o /dev/null
"/Applications/Arduino 3.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega1284 -DF_CPU=20000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -I/Users/BradJackson/Library/Arduino15/packages/MightyCore/hardware/avr/2.0.3/cores/MCUdude_corefiles -I/Users/BradJackson/Library/Arduino15/packages/MightyCore/har [... cut ...]
Using library ArduinoTapTempo-master at version 1.0 in folder: /Users/BradJackson/Google Drive/Jackson
exit status 1
Error compiling for board ATmega1284.

		BRAD JACKSON 
President / CEO

JACKSON AUDIO / JACKSON AMPWORKS

email: brad@jacksonaudio.net <mailto:brad@jacksonaudio.net> 
phone: 1-559-JACKSON (522-5766) 
web:  http://jackson.audio <http://jackson.audio/> 

110 Sports Parkway, Suite C
KELLER, TX 76248

 <https://facebook.com/jacksonaudio>	 <https://youtube.com/jacksonaudio>	 <https://instagram.com/jacksonaudio>	 <https://twitter.com/jackson_audio>

MCUdude, 2020-03-15:
I'm closing this issue because this isn't an issue related to MightyCore, but instead an incompatible library. If you want support for the 1284P you'll have to either do it yourself or contact the author of the library.

closed 2020-03-15 by MCUdude