MCUdude/MightyCore #168: Slow compilation

mrguen, 2019-11-11:
While searching for the reason of slow compilation for STM32 boards I found an issue that also concerns the MightyCore: files are archived again even thought they were not recompiled see https://forum.arduino.cc/index.php?topic=82527.msg4367805#msg4367805

MCUdude, 2019-11-16:
Well, I tested this both macOS and Windows. Both use archived files.

On macOS, first compilation:
```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/hans/Library/Arduino15/packages -hardware /Users/hans/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/hans/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/hans/Documents/Arduino/libraries -fqbn=MightyCore:avr:1284:bootloader=uart0,pinout=standard,variant=modelNonP,BOD=2v7,LTO=Os,clock=16MHz_external -ide-version=10810 -build-path /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387 -warnings=all -build-cache /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_cache_74871 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/hans/Library/Arduino15/packages -hardware /Users/hans/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/hans/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/hans/Documents/Arduino/libraries -fqbn=MightyCore:avr:1284:bootloader=uart0,pinout=standard,variant=modelNonP,BOD=2v7,LTO=Os,clock=16MHz_external -ide-version=10810 -build-path /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387 -warnings=all -build-cache /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_cache_74871 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
Using board '1284' from platform in folder: /Users/hans/Documents/Arduino/hardware/MightyCore/avr
Using core 'MCUdude_corefiles' from platform in folder: /Users/hans/Documents/Arduino/hardware/MightyCore/avr
Detecting libraries used...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -I/Users/hans/Documents/Arduino/hardware/MightyCore/avr/cores/MCUdude_corefiles -I/Users/hans/Documents/Arduino/hardware/MightyCore/avr/variants/standard /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -I/Users/hans/Documents/Arduino/hardware/MightyCore/avr/cores/MCUdude_corefiles -I/Users/hans/Documents/Arduino/hardware/MightyCore/avr/variants/standard /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387/sketch/Blink.ino.cpp -o /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387/preproc/ctags_target_for_gcc_minus_e.cpp
/Users/hans/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -Wextra -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -I/Users/hans/Documents/Arduino/hardware/MightyCore/avr/cores/MCUdude_corefiles -I/Users/hans/Documents/Arduino/hardware/MightyCore/avr/variants/standard /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387/sketch/Blink.ino.cpp -o /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387/sketch/Blink.ino.cpp.o
Compiling libraries...
Compiling core...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -x assembler-with-cpp -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -I/Users/hans/Documents/Arduino/hardware/MightyCore/avr/cores/MCUdude_corefiles -I/Users/hans/Documents/Arduino/hardware/MightyCore/avr/variants/standard /Users/hans/Documents/Arduino/hardware/MightyCore/avr/cores/MCUdude_corefiles/wiring_pulse.S -o /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387/core/wiring_pulse.S.o
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -I/Users/hans/Documents/Arduino/hardware/MightyCore/avr/cores/MCUdude_corefiles -I/Users/hans/Documents/Arduino/hardware/MightyCore/avr/variants/standard /Users/hans/Documents/Arduino/hardware/MightyCore/avr/cores/MCUdude_corefiles/wiring.c -o /var/folders/6l/ypg6qbw172v1s4vt [... cut ...]
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387/Blink.ino.elf /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387/Blink.ino.hex
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-size -A /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_420387/Blink.ino.elf
Sketch uses 1204 bytes (0%) of program storage space. Maximum is 130048 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 16375 bytes for local variables. Maximum is 16384 bytes.
```

mrguen, 2019-11-18:
Tested this morning

Windows 7 Pro 64 bits
Arduino 1.8.9
MightyCore 2.0.3

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\TG\AppData\Local\Arduino15\packages -hardware C:\Users\TG\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\TG\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\TG\Documents\Arduino\libraries -fqbn=MightyCore:avr:1284:bootloader=uart0,pinout=standard,variant=modelNonP,BOD=2v7,LTO=Os,clock=16MHz_external -ide-version=10809 -build-path C:\Users\TG\AppData\Local\Temp\arduino_build_406901 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\TG\AppData\Local\Arduino15\packages -hardware C:\Users\TG\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\TG\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\TG\Documents\Arduino\libraries -fqbn=MightyCore:avr:1284:bootloader=uart0,pinout=standard,variant=modelNonP,BOD=2v7,LTO=Os,clock=16MHz_external -ide-version=10809 -build-path C:\Users\TG\AppData\Local\Temp\arduino_build_406901 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino
Using board '1284' from platform in folder: C:\Users\TG\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.3
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\TG\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.3
WARNING: Category 'Language' in library ArduinoStreaming is not valid. Setting to 'Uncategorized'
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.3\\variants\\standard" "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_406901\\sketch\\Blink.ino.cpp" -o nul
Generating function prototypes...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.3\\variants\\standard" "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_406901\\sketch\\Blink.ino.cpp" -o "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_406901\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_406901\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.3\\variants\\standard" "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_406901\\sketch\\Blink.ino.cpp" -o "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_406901\\sketch\\Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.3\\variants\\standard" "C:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.3\\cores\\MCUdude_corefiles\\wiring_pulse.S" -o "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_406901\\core\\wiring_pulse.S.o"
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\TG\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.3\\variants\\standard" " [... cut ...]
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-objcopy" -O ihex -R .eeprom "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_406901/Blink.ino.elf" "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_406901/Blink.ino.hex"
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\TG\\AppData\\Local\\Temp\\arduino_build_406901/Blink.ino.elf"
Sketch uses 1214 bytes (0%) of program storage space. Maximum is 130048 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 16375 bytes for local variables. Maximum is 16384 bytes.

mrguen, 2019-11-25:
Surely there was a problem with my preferences file. The agressive caching feature was inactive. It had to be set as a preference in Arduino 1.8.9. 

Now in Arduino 1.8.10 the agressive caching is activated by default and can not be set as a preference in the Arduino IDE. 

I did a clean install of Arduino 1.8.10 and it does work as expected by compiling the core at first compilation and using it afterward.

closed 2019-11-25 by mrguen