MCUdude/MightyCore #169: MightyCore 324PB "INTERNAL2V56 was not declared in this scpe"

Dani-Hg, 2020-01-15:
Unfortunately it is not possible to set AREF to 1.1V or 2.56V. It is always on Vcc without an external capacitor.
There is always an error "INTERNAL2V56 was not declared in this scpe"
I ask for a solution.
(I switched from Atmega164A board to an Atmega 324PB board, with the Atmega164A board the program runs perfectly. I use Arduino 1.8.10 and MightyCore 2.0.3)

MCUdude, 2020-01-15:
Thanks for reporting. That is a very good catch!

This will be fixed soon. however, if you don't have time to wait you can add the ATmega324PB to this ifdef:
https://github.com/MCUdude/MightyCore/blob/14c6bec97e736cb32f64d2cefa9cdd55eda6e161/avr/cores/MCUdude_corefiles/Arduino.h#L96-L103

Dani-Hg, 2020-01-15:
Thank you for the quick answer, I added the entry and now the program can be compiled without errors. Unfortunately, it has no effect on the result. The PE4 (Aref PIN) is 0V and Vcc(5V) is still used as a reference. I tried analogReference (INTENAL), analogReference (INTENAL1V1), analogReference (INTENAL2V56) without changing anything.

Dani-Hg, 2020-01-15:
Update: it works partially, the internal Aref can be set correctly. Only no voltage is output at the AREF PIN. ????  (I tried BASCOM and then I measured 2.5V on the PE4). 
Thank you again

MCUdude, 2020-01-15:
There is no output voltage on the AREF pin. This pin is used if you're applying an external analog reference voltage.

Dani-Hg, 2020-01-15:
I don't actually need Aref Pin, I just connected a 100nF capacitor. (as it is in the data sheet). It is inexplicable to me how to separate the PIN from the AREF. I have now tested all possibilities with BASCOM (ADMUX = & B11000000 gives 2.53V to PE4, ADMUX = & B10000000 gives 1.09V to PE4, ADMUX = & B01000000 gives 5.0V to PE4). With Arduino I have 0V ??? 

As long as it works internally it is ok.

MCUdude, 2020-01-15:
Actually, twhen you set the analog reference voltage using `analogReference()`, you're actually not setting the reference. It's only storing the state:

https://github.com/MCUdude/MightyCore/blob/3da2e02281a4ce529f789ceefb7355514d930602/avr/cores/MCUdude_corefiles/wiring_analog.c#L36-L42

However, when you run `analogRead()`, you're setting the ADMUX register.
https://github.com/MCUdude/MightyCore/blob/3da2e02281a4ce529f789ceefb7355514d930602/avr/cores/MCUdude_corefiles/wiring_analog.c#L44-L96

MCUdude, 2020-03-15:
Fixed in the latest boards manager release, 2.0.4.

closed 2020-03-15 by MCUdude