MCUdude/MightyCore #171: COPY FLASH PAGES not working

leamendoza97, 2020-02-04:
Hi everybody, i'm not absolutly sure if this is a issue. I'm trying to use library ArduinoOTA and everything suggests it works. When the sketch is ready in the second half flash section, the library call copy flash pages function of optiboot and then the atmega reset (thanks of wdt). but boot old sketch!! i think the new sketch is not copied, but is in the second half flash section (i didn't check this because i dont know do it)

I'm testing in ATmega1284p. and i already tried compile mighty core with BIGBOOT=1 option (and burn it)

jandrassy, 2020-02-04:
Hi. I have [here](https://github.com/jandrassy/my_boards/tree/master/avr/bootloaders/optiboot8)a build of Optiboot with copy_flash_pages.

leamendoza97, 2020-02-04:
> Hi. I have [here](https://github.com/jandrassy/my_boards/tree/master/avr/bootloaders/optiboot8) a build of Optiboot with copy_flash_pages for 1284p

Thank you for your answer. Recently i try your hex but nothing changed.

leamendoza97, 2020-02-04:
I FOUND THE ISSUE.  When we compile with BIGBOOT or COPY_FLASH_PAGES the functions will include in the bootloader, BUT the call will not include beacuse in file https://github.com/MCUdude/MightyCore/blob/master/avr/bootloaders/optiboot_flash/optiboot_flash.c line 464 the precompiled directive COPY_FLASH_PAGES_FNC is WRONG, the real directive should be COPY_FLASH_PAGES (without _FNC).

jandrassy, 2020-02-06:
COPY_FLASH_PAGES_FNC was in my original pull request. but MCUdude made other changes in makefile and source code so merge was not possible and he only copied the function from my PR and renamed the macro.
@MCUdude Hans, could you fix it please? optiboot_flash.c line 464 COPY_FLASH_PAGES<s>_FNC</s>

MCUdude, 2020-02-06:
Fixed in the Optiboot flash repo:
https://github.com/MCUdude/optiboot_flash

MCUdude, 2020-03-15:
Fixed in the latest boards manager release, 2.0.4.

closed 2020-03-15 by MCUdude