MCUdude/MightyCore #173: abstract type 'awriter'

enzodello, 2020-02-13:
[Atmega1284P error.docx](https://github.com/MCUdude/MightyCore/files/4198630/Atmega1284P.error.docx)

 Hi i have this error when use awriter type.
If I Atmega2560 default core compilation is successful.

I tried to replace these files in yours cores, taking them from original arduino core and compilation is successful 👍 

HardwareSerial.h
HardwareSerial.cpp
Stream.h
Stream.cpp

MCUdude, 2020-02-13:
Hi!

Can you post your output error as plain text here instead of a docx document?

enzodello, 2020-02-13:
Arduino:1.8.11 (Windows 10), Scheda:"ATmega1284, Yes (UART0), Standard pinout, 1284P, BOD disabled, LTO disabled, External 16 MHz"

Zener_Diode_Tester:24:9: error: cannot declare variable 'n' to be of abstract type 'awriter'

 awriter n;

         ^

In file included from C:\Users\ig29143\Downloads\Zener_Diode_Tester\Zener_Diode_Tester.ino:22:0:

sketch\awriter.h:22:7: note:   because the following virtual functions are pure within 'awriter':

 class awriter: public Stream{

       ^~~~~~~

In file included from C:\Users\ig29143\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.3\cores\MCUdude_corefiles/HardwareSerial.h:29:0,

                 from C:\Users\ig29143\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.3\cores\MCUdude_corefiles/Arduino.h:288,

                 from sketch\Zener_Diode_Tester.ino.cpp:1:

C:\Users\ig29143\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.3\cores\MCUdude_corefiles/Stream.h:62:18: note: 	virtual void Stream::flush()

     virtual void flush() = 0;

                  ^~~~~

exit status 1
cannot declare variable 'n' to be of abstract type 'awriter'

MCUdude, 2020-02-13:
Can you provide a link or source to where I can find the source code? I have no idea what awriter is without it.

enzodello, 2020-02-13:
https://www.epemag3.com/lib/src/0320.zip
Zener diode project

MCUdude, 2020-02-13:
That zip file does not contain an Arduino project

enzodello, 2020-02-13:
sorry, i attached project in zip file
[Zener_Diode_Tester.zip](https://github.com/MCUdude/MightyCore/files/4198862/Zener_Diode_Tester.zip)

MCUdude, 2020-02-13:
Found it!
The fix will be applied in the next version of MightyCore. For now, you will have to remove the definition of the flush function in Stream.h:

https://github.com/MCUdude/MCUdude_corefiles/commit/74729f887f4a731d2a878468b016d049385cb2a8

closed 2020-02-13 by MCUdude