MCUdude/MightyCore #176: Use bootloader with a custom pinout.

Powertx, 2020-03-13:
Hello,
I'm so confused how this works. I have a custom board definition thing, that works with my arduino uno board, but i cant figure out how to use your bootloader with my boards.

Basically i want to use your atmega1284 bootloader, in my board definitons.
If i just add a custom arduino_pins.h in your boards folder it works fine. I just cant get it to work with my boards download.

Can anyone help me? Hope this makes sense...

Powertx, 2020-03-13:
I keep getting
"recipe.preproc.macros pattern is missing" when trying yo compile...

MCUdude, 2020-03-14:
You should fork MightyCore and add your own variant. Use the `standard` one as a template. After you've done this you'll just use this version in the boards.txt file, just like I've done here:

https://github.com/MCUdude/MightyCore/blob/703121873af406dfa7a58f3df4db21b51c8005e9/avr/boards.txt#L73

If you're trying to create your own Arduino core, just like MightyCore is, I can't help you with that. The code for this project is open-source, and you're free to figure out the details.

closed 2020-03-14 by MCUdude

jandrassy, 2020-03-15:
https://arduino.stackexchange.com/questions/54484/adding-a-custom-board-to-the-arduino-ide/60660#60660

Powertx, 2020-03-16:
Thanks, for help. I figured i only need your bootloader(hex file). Everything else is just arduino stuff.