MCUdude/MightyCore #177: Error during build: fork/exec /bin/avr-g++: no such file or directory

rejoys-servify, 2020-03-18:
I am trying to compile sketch for Atmeg16A which is defined in MightyCore, using Arduino CLI.
Everything works fine when I do this on my mac, but throws this "Error during build: fork/exec /bin/avr-g++: no such file or directory" when I try to do the same in my raspberry pi. I am able to compile sketch for builtin boards like uno and mega on my pi. Similar issue are on Arduino forum but for Arduino IDE and the suggested solution I found was to uninstall and reinstall the board files, the IDE itself etc. Any help will be grateful. Thanks in advance.
[image]

gferon, 2020-03-29:
It looks like you just need to install `arduino:avr` as well. Well, at least it worked for me.

ioExpander, 2020-04-19:
Same here when running on the arduino package from Arch Linux. It does not bundle Arduino:avr but uses upstream tools. This setup confuses MightyCore.
**I solved it by going to Tools>boards>Board Manager and installing  the latest "Arduino AVR Boards".** 
At least this fixes the issue.
So it seems that the MightyCore package depends on the default Arduino AVR tools. I do not know whether this is good practice.

MCUdude, 2020-04-19:
The toolchain is usually installed in together with the IDE. However, like @ioExpander mentioned, you can install the Arduino AVR Boards from boards manager if you're having issues.

closed 2020-04-19 by MCUdude