MCUdude/MightyCore #179: Support of the W5500 chipset (Ethernet doesn't work)

nofei, 2020-04-01:
Hello,

my name is Norbert and I have problems with the ethernet connection to a ATmega644.
My setup:
# ATmega644 with the MightyCore (burned with the AVRISP MkII, no bootloader)
# Ethernetboard with an W5500 chipset connect via SPI

I tried some example code (WebServer from GitHub) with the Ethernet.h and Ethernet2.h --> no success.
The Ethernetboard is working, because I use that with an Arduino NANO. After some research in the Internet I know, that the W5500 only works with the ethernet2.h library.

My question to you: Does the MightyCore support the W5500 chipset?? 

Thanks for your help.

MCUdude, 2020-04-01:
See #72 and #125. The W5500 is supported, I have tested that exact chip myself (using a WIZ850io board). Make sure you use the correct IO pins on the ATmega644. The pinout is different compared to an Arduino Nano.

nofei, 2020-04-01:
Thanks for the replay. 
I did the pinout correct, because I used the description ATmega644 (standard) from GitHub. Could it be, that there are differences between the ethernet.h and ethernet2.h libs?

jandrassy, 2020-04-01:
Ethernet2 is obsolete. Ethernet library  supports W5500 from august 2018

MCUdude, 2020-05-15:
I'm closing this issue because It's not related to MightyCore really. If you need help with your project I suggest you ask for help over at the Arduino forums.

closed 2020-05-15 by MCUdude