MCUdude/MightyCore #181: Atmega32 Serial.readStringuntil("/n") error

AKASHWORLD614, 2020-05-04:
Hello
I simply keep getting this error during compilation 
File1:90:43: error: invalid conversion from 'const char*' to 'char' [-fpermissive]
     idStatus = Serial.readStringUntil("/n");
                                           ^
and this works with atmega328p but the pins were not sufficient so moved to a atmega32 
please help me out with this error as fast as possible

MCUdude, 2020-05-04:
You are trying to convert from const char* to char, which MightyCore won't let you do.

I may be able to help if you post your code.

MCUdude, 2020-05-04:
But I'm pretty sure `idStatus` has to be the datatype `String`, like so:

```c++
String idStatus = Serial.readStringUntil('\n');
```

AKASHWORLD614, 2020-05-04:
> But I'm pretty sure `idStatus` has to be the datatype `String`, like so:
> 
> ```c++
> String idStatus = Serial.readStringUntil('\n');
> ```

I have used the String idStatus; in the start of the code before setup
 i am just trying to replicate arduino uno project with atmega32

AKASHWORLD614, 2020-05-04:
> But I'm pretty sure `idStatus` has to be the datatype `String`, like so:
> 
> ```c++
> String idStatus = Serial.readStringUntil('\n');
> ```

I have solved the problem just replaced
Serial.readStringuntil("\n");
with 
Serial.readStringuntil('\n');

closed 2020-05-04 by AKASHWORLD614

MCUdude, 2020-05-04:
Ah, the quotes. Great to hear it's working for you!

AKASHWORLD614, 2020-05-04:
> Ah, the quotes. Great to hear it's working for you!

Is it possible to add a MFRC522 RFID Reader to the Atmega32 and the chip keeps blinking the led on PB0 even if there is no bootloader but code works only after power interrupt