MCUdude/MightyCore #183: Custom board compilation error "Arduino.h: No such file or directory"

mrguen, 2020-05-19:
Hello,
When creating a custom board under the hardware folder, I get the compilation error "Arduino.h: No such file or directory"

The configuration is 
Arduino 1.8.12
adding 
* board.txt in the "avr" architecture folder
* pins_arduino.h in the variants/Narrow folder
and that is all.

Here is the boards.txt content:
`###########################################################################################################
## This file is referencing 644 Narrow and 1284 Narrow boards. 																					 ##
## It uses the MightyCore                                               																 ##
## https://github.com/MCUdude/MightyCore                                                                 ##
##                                                                                                       ##
## This is the file where fuse settings, clock frequencies, upload ports, bootloaders and                ##
## compiler flags are located.                                                                           ##
##                                                                                                       ##
##                                           |                                                           ##
###########################################################################################################

# Menu options
menu.clock=Clock
menu.BOD=BOD
menu.LTO=Compiler LTO
menu.variant=Variant
menu.pinout=Pinout
menu.bootloader=Bootloader

###########################
#### 644 Narrow  			 ####
###########################

# General
644_Narrow.name=644 Narrow
644_Narrow.upload.tool=arduino:avrdude
644_Narrow.upload.maximum_data_size=4096
644_Narrow.bootloader.tool=arduino:avrdude
644_Narrow.bootloader.unlock_bits=0x3f
644_Narrow.bootloader.lock_bits=0x0f
644_Narrow.build.core=arduino:MCUdude_corefiles
644_Narrow.build.board=AVR_ATmega644
644_Narrow.build.extra_flags={build.clkpr}

# Upload port select
644_Narrow.menu.bootloader.uart0=Yes (UART0)
644_Narrow.menu.bootloader.uart0.upload.maximum_size=64512
644_Narrow.menu.bootloader.uart0.upload.protocol=arduino
644_Narrow.menu.bootloader.uart0.upload.port=UART0
644_Narrow.menu.bootloader.uart0.bootloader.high_fuses=0xd6
644_Narrow.menu.bootloader.uart0.bootloader.file=optiboot_flash/bootloaders/{build.mcu}/{build.clock_speed}/optiboot_flash_{build.mcu}_{upload.port}_{upload.speed}_{build.clock_speed}_{build.bootloader_led}_BIGBOOT.hex

644_Narrow.menu.bootloader.no_bootloader=No bootloader
644_Narrow.menu.bootloader.no_bootloader.upload.maximum_size=65536
644_Narrow.menu.bootloader.no_bootloader.bootloader.high_fuses=0xd7
644_Narrow.menu.bootloader.no_bootloader.bootloader.file=empty/empty.hex

# Pinouts
644_Narrow.menu.pinout.standard=Standard pinout
644_Narrow.menu.pinout.standard.build.variant=Narrow
644_Narrow.menu.pinout.standard.build.bootloader_led=B7

# Variants
644_Narrow.menu.variant.modelP=644P / 644PA
644_Narrow.menu.variant.modelP.build.mcu=atmega644p

# Brown out detection
644_Narrow.menu.BOD.2v7=BOD 2.7V
644_Narrow.menu.BOD.2v7.bootloader.extended_fuses=0xfd
644_Narrow.menu.BOD.4v3=BOD 4.3V
644_Narrow.menu.BOD.4v3.bootloader.extended_fuses=0xfc
644_Narrow.menu.BOD.1v8=BOD 1.8V
644_Narrow.menu.BOD.1v8.bootloader.extended_fuses=0xfe
644_Narrow.menu.BOD.disabled=BOD disabled
644_Narrow.menu.BOD.disabled.bootloader.extended_fuses=0xff

# Compiler link time optimization
644_Narrow.menu.LTO.Os=Disabled
644_Narrow.menu.LTO.Os.compiler.c.extra_flags=
644_Narrow.menu.LTO.Os.compiler.c.elf.extra_flags=
644_Narrow.menu.LTO.Os.compiler.cpp.extra_flags=
644_Narrow.menu.LTO.Os.ltoarcmd=avr-ar

644_Narrow.menu.LTO.Os_flto=Enabled
644_Narrow.menu.LTO.Os_flto.compiler.c.extra_flags=-Wextra -flto -g
644_Narrow.menu.LTO.Os_flto.compiler.c.elf.extra_flags=-w -flto -g
644_Narrow.menu.LTO.Os_flto.compiler.cpp.extra_flags=-Wextra -flto -g
644_Narrow.menu.LTO.Os_flto.ltoarcmd=avr-gcc-ar

# Clock frequencies
644_Narrow.menu.clock.16MHz_external=External 16 MHz
644_Narrow.menu.clock.16MHz_external.upload.speed=115200
644_Narrow.menu.clock.16MHz_external.bootloader.low_fuses=0xf7
644_Narrow.menu.clock.16MHz_external.build.clkpr=
644_Narrow.menu.clock.16MHz_external.build.clock_speed={build.f_cpu}
644_Narrow.menu.clock.16MHz_external.build.f_cpu=16000000L

644_Narrow.menu.clock.8MHz_internal=Internal 8 MHz
644_Narrow.menu.clock.8MHz_internal.upload.speed=38400
644_Narrow.menu.clock.8MHz_internal.bootloader.low_fuses=0xe2
644_Narrow.menu.clock.8MHz_internal.build.clkpr=
644_Narrow.menu.clock.8MHz_internal.build.clock_speed={build.f_cpu}
644_Narrow.menu.clock.8MHz_internal.build.f_cpu=8000000L

644_Narrow.menu.clock.4MHz_internal=Internal 4 MHz
644_Narrow.menu.clock.4MHz_internal.upload.speed=9600
644_Narrow.menu.clock.4MHz_internal.bootloader.low_fuses=0xe2
644_Narrow.menu.clock.4MHz_internal.build.clkpr=-DOSC_PRESCALER=0x01
644_Narrow.menu.clock.4MHz_internal.build.clock_speed=8000000L
644_Narrow.menu.clock.4MHz_internal.build.f_cpu=4000000L

644_Narrow.menu.clock.2MHz_internal=Internal 2 MHz
644_Narrow.menu.clock.2MHz_internal.upload.speed=9600
644_Narrow.menu.clock.2MHz_internal.bootloader.low_fuses=0xe2
644_Narrow.menu.clock.2MHz_internal.build.clkpr=-DOSC_PRESCALER=0x02
644_Narrow.menu.clock.2MHz_internal.build.clock_speed=8000000L
644_Narrow.menu.clock.2MHz_internal.build.f_cpu=2000000L

644_Narrow.menu.clock.1MHz_internal=Internal 1 MHz
644_Narrow.menu.clock.1MHz_internal.upload.speed=9600
644_Narrow.menu.clock.1MHz_internal.bootloader.low_fuses=0x62
644_Narrow.menu.clock.1MHz_internal.build.clkpr=
644_Narrow.menu.clock.1MHz_internal.build.clock_speed={build.f_cpu}
644_Narrow.menu.clock.1MHz_internal.build.f_cpu=1000000L

`

MCUdude, 2020-05-19:
I'm not sure why you have added `arduino:` in front of `upload.tool`, `bootloader.tool` and `build.tool`. Try to remove this.

mrguen, 2020-05-19:
@MCUdude  because it is in the User/Arduino/hardware folder: there isn't any core resourses there. All is located in AppData/Arduino15. So without the prefix, it does not find the compiler an upload tools etc

MCUdude, 2020-05-19:
Do you don't provide the entire core, just the boards.txt and some other few files?

I suggest you start out by modifying an existing MightyCore installation that works and add your new board entry to the list of boards in the existing boards.txt. When you got everything working, then you can start removing stuff you don't need. Also, if you're modifying files, it's much better to do a manual install rather than boards manager.

MCUdude, 2020-06-21:
Closing due to not really a MightyCore related issue

closed 2020-06-21 by MCUdude