MCUdude/MightyCore #184: Suggestion - for parts with weird names for USART registers, but the same function

SpenceKonde, 2020-05-24:
On those parts, it would make life easier if you provided #defines for the standard names, which - unless I'm mistaken - I don't think you currently do (ie, for the atmega16 which doesn't have the 0 in it's USART registers, compile a list of them, and #define (registername with 0) (registername without 0). The only tricky one is that one register that has the high-bit nonsense.... I did this in jtag2updi so I could clean up the code everywhere else and get rid of the ATmega16 special cases that littered it's USART code.

MCUdude, 2020-05-27:
That's a good idea, why haven't I thought of this before? I'll see what I can do!

MCUdude, 2020-05-27:
I just realized, [I wrote this guide a few years ago](https://github.com/MCUdude/MightyCore/blob/master/Library_porting.md). This could could just be copied into pins_arduino.h and many/most libraries will just work.

> The only tricky one is that one register that has the high-bit nonsense

What register? Would you like to pass me "a pointer" to it? 😉

SpenceKonde, 2020-05-27:
Atmega16 UCSRC I think it is

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

closed 2022-12-27 by MCUdude