MCUdude/MightyCore #186: Use 7.3.0-atmel3.6.1-arduino7 to fix power.h for 324PB

SpenceKonde, 2020-07-07:
Older versions of toolchain's power.h does not support the 324PB - this was one of the patches I got into that version!

MCUdude, 2021-02-19:
7.3.0-atmel3.6.1-arduino8 is used if you install the latest boards manager release.

closed 2021-02-19 by MCUdude

SpenceKonde, 2021-02-19:
Do you mean 7.3.0-atmel3.6.1-arduino7 or is there a new one? My boards manager isn't showing any 7.3.0-atmel3.6.1-arduino8 in the official package index

MCUdude, 2021-02-19:
Yes, I mean *arduino7. I mixed it up with avrdude-arduino18.

SpenceKonde, 2021-02-19:
I was excited for a minute there :-P

SpenceKonde, 2021-02-19:
(the azduino3 toolchain package is running a little over $10/month currently just for bandwidth)

MCUdude, 2021-02-19:
Whoa! I should put up a script that constantly downloads and delete the downloaded archive to help you out then 😆