MCUdude/MightyCore #188: Are pin-change interrupts (PCINT) supported in MightyCore?

bgoolsby, 2020-08-19:
I've been using the Mega644P and find that only INT0-2 are supported.  Am I correct in this, and are there plans to support the PCINT pins?
Thanks,

MCUdude, 2020-08-19:
You're free to use PCINT, but they aren't supported so that you can use `attachInterrupt`. You'll have to use them directly. There's plenty of tutorials on the internet on how you can do this. There also exists an Arduino PCINT library out there I believe.

There are no plans to add PCINT support, as this is something the official Arduino cores don't support either.

closed 2020-08-19 by MCUdude

bgoolsby, 2020-08-19:
Thanks, I'll look this up.

bgoolsby, 2020-08-20:
Was this the proper venue to ask my question?  I didn't find a forum
dedicated to Mightycore.