MCUdude/MightyCore #194: Latest release no exporting compiled binary with bootloader

glyndavidson, 2020-11-02:
Just updated to Arduino IDE 1.8.13 and MightyCore 2.0.6
It used to be the case that when I exported the compiled binary, I would get two hex files, one with bootloader and one without.
Now, I'm only getting the none bootloader version.

If I turn on verbose compiling, I can see this command:
`C:\Users\Glytc\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.6/scripts/delete_merged_output.bat true C:\Users\Glytc\AppData\Local\Temp\arduino_build_796178\sketch.ino.with_bootloader.hex `

If I open up the delete_merged_output.bat file and comment out the line
`if "%1" == "false" del "%2"`
Then I would expect to find this file:
C:\Users\Glytc\AppData\Local\Temp\arduino_build_796178\sketch.ino.with_bootloader.hex 
Right?
But it's not there.

**Board**:              ATMega1284
**Clock**:              External 12Mhz
**Bod**:                Bod Disabled
**EEPROM**:         EEPROM Retained
**Compiler LTO**: LTO Disabled
**Variant**:           1284p
**Pinout**:            Standard Pinout
**Bootloader**:    Yes (UART0)
**Port**:                COM6

MCUdude, 2020-11-02:
This is a windows related issue and was fixed in d869a50. Expect a new release soon. Meanwhile, you can just look at the Arduino output to find out where the hex file is originally stored. Just turn on verbose output in the IDE settings.

glyndavidson, 2020-11-02:
Thanks for the reply. I've tried that already.
According to the verbose output, the file is stored in:
C:\Users\Glytc\AppData\Local\Temp\arduino_build_796178\sketch.ino.with_bootloader.hex
I've looked, but it's not there. 
As I say, have also commented out the line in the .bat file which deletes the temp file, but it's still not there.

glyndavidson, 2020-11-02:
I just merged the changes in d869a50 (in my windows environment) but still have the same problem.

glyndavidson, 2020-11-02:
Verbose output:

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Glytc\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Glytc\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=MightyCore:avr:1284:bootloader=uart0,pinout=standard,eeprom=keep,variant=modelP,BOD=disabled,LTO=Os,clock=12MHz_external -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\Glytc\AppData\Local\Temp\arduino_build_83433 -warnings=default -build-cache C:\Users\Glytc\AppData\Local\Temp\arduino_cache_896818 -prefs=build.warn_data_percentage=75 -verbose D:\Desktop\vario\sketch\sketch.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Glytc\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Glytc\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=MightyCore:avr:1284:bootloader=uart0,pinout=standard,eeprom=keep,variant=modelP,BOD=disabled,LTO=Os,clock=12MHz_external -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\Glytc\AppData\Local\Temp\arduino_build_83433 -warnings=default -build-cache C:\Users\Glytc\AppData\Local\Temp\arduino_cache_896818 -prefs=build.warn_data_percentage=75 -verbose D:\Desktop\vario\sketch\sketch.ino
Using board '1284' from platform in folder: C:\Users\Glytc\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.6
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\Glytc\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.6

[... resolving libraries ...]
[... generating prototypes ...]
[... compiling sketch ...]
[... compiling libraries ...]
[... compiling core ...]

Linking everything together...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -Os -Wl,--gc-sections,--section-start=.FAR_MEM1=0x10000 -mmcu=atmega1284p -o "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433/sketch.ino.elf" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433\\sketch\\BMP388_DEV.cpp.o" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433\\sketch\\Device.cpp.o" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433\\sketch\\LIS3DH_Accel.cpp.o" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433\\sketch\\SSD1306_DISPLAY.cpp.o" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433\\sketch\\SparkFun_RV8803.cpp.o" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433\\sketch\\sketch.ino.cpp.o" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433\\libraries\\Wire\\Wire.cpp.o" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433\\libraries\\Wire\\utility\\twi.c.o" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433\\libraries\\SPI\\SPI.cpp.o" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433/core\\core.a" "-LC:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433" -lm
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433/sketch.ino.elf" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433/sketch.ino.eep"
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-objcopy" -O ihex -R .eeprom "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433/sketch.ino.elf" "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433/sketch.ino.hex"
cmd /C "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433/sketch.ino.elf" > "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433/sketch.ino_atmega1284p_12000000L.lst"
Using library Wire at version 1.0 in folder: C:\Users\Glytc\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.6\libraries\Wire 
Using library EEPROM at version 2.0 in folder: C:\Users\Glytc\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.6\libraries\EEPROM 
Using library SPI at version 1.0 in folder: C:\Users\Glytc\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.6\libraries\SPI 
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\Glytc\\AppData\\Local\\Temp\\arduino_build_83433/sketch.ino.elf"
Sketch uses 67014 bytes (51%) of program storage space. Maximum is 130048 bytes.
Global variables use 6778 bytes (41%) of dynamic memory, leaving 9606 bytes for local variables. Maximum is 16384 bytes.
C:\Users\Glytc\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.6/scripts/delete_merged_output.bat true C:\Users\Glytc\AppData\Local\Temp\arduino_build_83433\sketch.ino.with_bootloader.hex 

D:\Desktop\vario\sketch>rem if "true" == "false" del "C:\Users\Glytc\AppData\Local\Temp\arduino_build_83433\sketch.ino.with_bootloader.hex" 
cmd /C copy C:\Users\Glytc\AppData\Local\Temp\arduino_build_83433\sketch.ino_atmega1284p_12000000L.lst D:\Desktop\vario\sketch 
        1 file(s) copied.

```

glyndavidson, 2020-11-02:
Contents of C:\Users\Glytc\AppData\Local\Temp\arduino_build_83433\
[image]

MCUdude, 2020-11-02:
I just pulled the latest version of MightyCore on my Windows computer. I compiled a sketch, saved it and ran Export compiled binary. Here's the output:

[image]

glyndavidson, 2020-11-02:
Just worked for me too with an empty sketch. I'll see if I can narrow down the problem...

glyndavidson, 2020-11-02:
It's working now but I'm afraid I can't tell you what the problem was.

It's a big file, currently 8042 lines of code (I know, I know, I need to break it into smaller files) with a number of libraries.
I narrowed the problem down to a function call from the main loop.
Commented out all of the functions and it exported the bootloader version.

Uncommented them, problem persisted.
Further narrowed it down to one switch statement which calls different functions in different conditions.
Comment out the switch statement and I get the bootloader version.
One by one, I uncommented the function calls in the switch statement.
The problem didn't happen again though. With all functions uncommented it compiled the boot loader version.

I've done a diff check between my original version and the version which gives me the bootloader and there are no differences. Very strange.

glyndavidson, 2020-11-02:
So I went back to my original file and still had the same problem. 
I've been able to narrow it down to several functions which do little more than this:

```
void changeSettingX(void){
  if(x){
   x = false;
  }
  else{
   x = true;
  }
 EEPROM.write(ADDR, x);
}
```

Luckily, they're all called by one line in the main loop. So if I comment out that line, it exports with the bootloader just fine.
Then, if I uncomment it, it still exports the bootloader.

However, if I close the IDE, then open it again, the same thing happens. I have to comment out that one line. Export. Uncomment it. And Export again.

So, I've been able to find a work around, but I can't find anything that's causing the problem.

MCUdude, 2020-11-17:
@glyndavidson can you try with the previous MightyCore release (v2.0.5) and see if it makes a difference?

jandrassy, 2020-11-17:
could it be this?
https://github.com/arduino/arduino-cli/pull/1017

MCUdude, 2020-11-17:
> could it be this?
arduino/arduino-cli#1017

It may be? I'm not into Go or the arduino-cli project, so I can't tell what changes that commit brought.

@glyndavidson does it work if you export the compiled binary without bootloader? If so, I'm pretty sure @jandrassy is correct.

MCUdude, 2020-12-02:
v2.0.7 is available through boards manager.

closed 2020-12-02 by MCUdude