MCUdude/MightyCore #195: 'c:\Program' is not recognized as an internal or external command, operable command or batch file

Yohan-Gunawardena, 2020-11-14:
There is a problem with version 2.0.6. I got the following error message when I try to compile for Atmega32A; 
"'C:\Program' is not recognized as an internal or external command,
operable program or batch file.
exit status 1
Error compiling for board ATmega32."
[image]

The problem fix after I install the old version; 2.0.5
[image]

MCUdude, 2020-11-14:
Is the problem gone if you do an manual install (remove the boards manager install first)? I pushed a fix recently, but I've not pushed a release yet.

Yohan-Gunawardena, 2020-11-14:
I used the link to install the board. Not the zip file.
[image]

MCUdude, 2020-11-14:
I'm asking you if you can uninstall the boards manager version to try the manual install version. I believe the bug is gone in the manual install version, but it would be nice if you could test it just to be sure.

If this solves your issue, I'll release v2.0.7 as soon as I can.

MCUdude, 2020-12-02:
2.0.7 is available through boards manager

closed 2020-12-02 by MCUdude