MCUdude/MightyCore #196: Atmega1284P restarting when execute optiboot_writePage

castrofernando, 2020-11-16:
My board restarts everytime i try to call the function optiboot_writePage. To do some tests, i just upload the SerialReadWrite.ino from example folder and same issue happens.

Can anyone give me some tip?

Tks

MCUdude, 2020-11-16:
It is a while since I've tried this example, so I grabbed my DIP-40 board, inserted an ATmega1284P in it, flashed a 16 MHz bootloader, and loaded the example.

Everything works fine here. The microcontroller doesn't reset, and the written content is preserved through a power cycle. You are using a bootloader, right?

castrofernando, 2020-11-16:
Actually, maybe this should be my mistake.I also tried ArduinoOTA, that also use same optiboot and got same problem when write. I'm didn't using bootloader, i'm flashing the example directly using ISP with USBasp. Do you have any other recomendation? I'm using 3v3 power supply and 16Mhz, flashing using AVRDudes 2.13

castrofernando, 2020-11-17:
Thank you @MCUdude! the problem was i'm not using bootloader. Now i was able to flash the bootloader correctly and worked perfectly. Thank you again!

closed 2020-11-17 by castrofernando