MCUdude/MightyCore #197: Atmega1284p -> verification error, first mismatch at byte 0x0d00 - 0x88 != 0xa8

castrofernando, 2020-11-17:
I'm trying to use the bootloader _optiboot_flash_atmega1284p_UART1_115200_16000000L_B0_BIGBOOT.hex_ , but i'm getting problem in application data after second flash.
Not sure if it's a issue or i'm missing something.

1. I've flashed the bootloader using platformio.
2. Flashed my program data once. (Everything works properly this time... When i turn on the board, the bootload start and then jump to my program, that works perfectly blinking a led)
3. Flashed my program second time, adding one more led to blink ( or any change i do). Then, program always give an error -> verification error, first mismatch at byte 0x0d00 - 0x88 != 0xa8. 
The bootloader keeps working, seems in second time it write in a wrong memory position.

Below i share the verbose:
1) Burning the bootloader
2) Burning flash first time and working ok - (tried with uart1 and usbasp - same result) 
2) Burning flash second time and main program stop work. only bootloader keeps ok - (tried with uart1 and usbasp - same result) 

**1) BURNING THE BOOTLOADER:**

`> Executing task: C:\Users\Administrador\.platformio\penv\Scripts\pio.exe run --target bootloader --environment MiniCore <

Processing MiniCore (platform: atmelavr; board: ATmega1284P; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega1284P.html
PLATFORM: Atmel AVR (3.0.0) > ATmega1284P
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 127KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr-mightycore 2.0.6
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 19 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Ethernet> 0.0.0-alpha+sha.9f41e8231b
|   |-- <SPI> 1.0
|-- <RTClib> 1.12.1
|   |-- <Wire> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <Optiboot flash read/write> 1.0
|-- <Yet Another Arduino Wiegand Library> 2.0.0
|-- <Yet Another Arduino PcInt Library> 2.1.0
Building in release mode

TARGET CONFIGURATION:
---------------------
Target = atmega1284p
Clock speed = 16000000L
Oscillator = external
BOD level = 2.7v
Save EEPROM = yes
UART port = uart1
Clock output = no
JTAG enable = no
---------------------

Selected fuses: [lfuse = 0xf7, hfuse = 0xd6, efuse = 0xfd]
Setting fuses

avrdude: Version 6.3, compiled on Sep 12 2016 at 17:24:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Administrador\.platformio\packages\tool-avrdude\avrdude.conf"

         Using Port                    : USB
         Using Programmer              : usbasp
avrdude: Warning: cannot open USB device: Function not implemented
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
[... 412 lines cut ...]

Writing | ################################################## | 100% 4.20s

avrdude: 6744 bytes of flash written
avrdude: verifying flash memory against .pio\build\MiniCore\firmware.hex:
avrdude: load data flash data from input file .pio\build\MiniCore\firmware.hex:
avrdude: input file .pio\build\MiniCore\firmware.hex contains 6744 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 3.55s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0d00
         0x88 != 0xa8
avrdude: verification error; content mismatch

avrdude: safemode: hfuse reads as D6
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D6, L:F7)

avrdude done.  Thank you.

*** [upload] Error 1
========================================================== [FAILED] Took 9.13 seconds ==========================================================
The terminal process "C:\Users\Administrador\.platformio\penv\Scripts\pio.exe 'run', '--target', 'upload'" terminated with exit code: 1.`

MCUdude, 2020-11-17:
I'll give you a better answer later this evening, but it is most likely because the (avrdude) -D flag is added. Can you post your platformio.ini config?

Oh, I took the liberty to edit your original post to make it more readable! Feel free to steal the tricks! 😉

castrofernando, 2020-11-17:
Got your tricks =]  Tks!!

I did some more tests, and then seems using UART1 is working now.... not sure what i did.
Using USBASP still happening same problem.

**platform.ini**

<details>
```
[env:MiniCore]
platform = atmelavr
board = ATmega1284P

; change microcontroller
board_build.mcu = atmega1284p
board_build.f_cpu = 16000000L
framework = arduino

board_hardware.oscillator = external ; Oscillator type
board_hardware.uart = uart1          ; Set UART to use for serial upload
board_bootloader.speed = 115200      ; Set bootloader baud rate
board_hardware.bod = disabled           ; Set brown-out detection
board_hardware.eesave = yes          ; Preserve EEPROM when uploading using programmer

; Usar para programação ISP usbasp
;upload_protocol = usbasp             ; Use the USBasp as programmer
;upload_flags =                       ; Select USB as upload port and divide the SPI clock by 8
;  -PUSB
;  -v

lib_ignore = TinyWireM
lib_deps =  
  paulstoffregen/Ethernet @ 0.0.0-alpha+sha.9f41e8231b

```

MCUdude, 2020-11-17:
> Got your tricks =] Tks!!

Awesome! However, markdown is sometimes very unforgiving, so I actually added a blank line between the "details" and the three ```'s. Never understood why it is so picky...

The reason why it fails with the USBasp programmer is that the upload command that's used is actually not meant to be used with dedicated programmers, but rather the bootloader, which has requires the -D flag to be set.

Try this configuration instead. I've worked together with the PlatformIO developers to streamline the process even more:

<details>

```ini
[platformio]
default_envs = Upload_UART ; Default build target

; Common settings for all environments
[env]
platform = atmelavr
framework = arduino

; TARGET SETTINGS
; Chip in use
board = ATmega1284P
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L
; Current pinout
board_build.variant = standard

; BUILD OPTIONS
; Comment out to enable LTO (this line unflags it)
build_unflags = -flto
; Extra build flags
build_flags = 

; SERIAL MONITOR OPTIONS
; Serial monitor port defined in the Upload_UART environment
monitor_port = ${env:Upload_UART.upload_port}
; Serial monitor baud rate
monitor_speed = 115200

; Run the following command to upload with this environment
; pio run -e Upload_UART -t upload
[env:Upload_UART]
;  Serial bootloader protocol
upload_protocol = arduino
; Serial upload port
upload_port =
; Get upload baud rate defined in the fuses_bootloader environment
board_upload.speed = ${env:fuses_bootloader.board_bootloader.speed}

; Run the following command to upload with this environment
; pio run -e Upload_ISP -t upload
[env:Upload_ISP]
; Custom upload procedure
upload_protocol = custom
; Avrdude upload flags
upload_flags =
  -C$PROJECT_PACKAGES_DIR/tool-avrdude/avrdude.conf
  -p$BOARD_MCU
  -PUSB
  -cusbasp
; Avrdude upload command
upload_command = avrdude $UPLOAD_FLAGS -U flash:w:$SOURCE:i

; Run the following command to set fuses
; pio run -e fuses_bootloader -t fuses
; Run the following command to set fuses + burn bootloader
; pio run -e fuses_bootloader -t bootloader
[env:fuses_bootloader]
board_hardware.oscillator = external ; Oscillator type
board_hardware.uart = uart1          ; Set UART to use for serial upload
board_bootloader.speed = 115200      ; Set bootloader baud rate
board_hardware.bod = 2.7v            ; Set brown-out detection
board_hardware.eesave = yes          ; Preserve EEPROM when uploading using programmer
upload_protocol = usbasp             ; Use the USBasp as programmer
upload_flags =                       ; Select USB as upload port and divide the SPI clock by 8
  -PUSB
  -B8
```
</details>

The idea is that upload using programmer, upload using a USB to serial adapter, and set fuses/bootloader are now split into separate environments. If you're using VSCode you can just select the environment you want to use to upload, and it will just work! However, you'll still have to set fuses/burn bootloader by executing a command:  
`pio run --target bootloader --environment  fuses_bootloader`.

MCUdude, 2020-11-21:
Did this solve your issue?

castrofernando, 2020-11-21:
Sorry for delay. I tried once and got same problem, but i need to try more, maybe i did something wrong. i'll try more later and let the feedback here.
Once i be able to do it work, i'll try to make it work with a ethernet upload ( this is my target, using optiboot... but still cannot arrive on it). For now, i can say that Serial worked properly.

Sercurio, 2022-09-14:
Same problem here.
I've got the same configuration on two projects. Upload works on one of them but no on the other.

platform.ini
```
[platformio]
default_envs = Upload_UART ; Default build target

; Common settings for all environments
[env]
platform = atmelavr
framework = arduino

; TARGET SETTINGS
; Chip in use
board = ATmega1284P
; Clock frequency in [Hz]
board_build.f_cpu = 16000000L
; Current pinout
board_build.variant = standard

; BUILD OPTIONS
; Comment out to enable LTO (this line unflags it)
build_unflags = -flto
; Extra build flags
build_flags = 

; SERIAL MONITOR OPTIONS
; Serial monitor port defined in the Upload_UART environment
monitor_port = ${env:Upload_UART.upload_port}
; Serial monitor baud rate
monitor_speed = 115200

; Run the following command to upload with this environment
; pio run -e Upload_UART -t upload
[env:Upload_UART]
;  Serial bootloader protocol
upload_protocol = arduino
; Serial upload port
upload_port =
; Get upload baud rate defined in the fuses_bootloader environment
board_upload.speed = ${env:fuses_bootloader.board_bootloader.speed}

; Run the following command to upload with this environment
; pio run -e Upload_ISP -t upload
[env:Upload_ISP]
; Custom upload procedure
upload_protocol = custom
; Avrdude upload flags
upload_flags =
  -C$PROJECT_PACKAGES_DIR/tool-avrdude/avrdude.conf
  -p$BOARD_MCU
  -PUSB
  -cusbasp
; Avrdude upload command
upload_command = avrdude $UPLOAD_FLAGS -U flash:w:$SOURCE:i

; Run the following command to set fuses
; pio run -e fuses_bootloader -t fuses
; Run the following command to set fuses + burn bootloader
; pio run -e fuses_bootloader -t bootloader
[env:fuses_bootloader]
board_hardware.oscillator = external ; Oscillator type
board_hardware.uart = uart1          ; Set UART to use for serial upload
board_bootloader.speed = 115200      ; Set bootloader baud rate
board_hardware.bod = 2.7v            ; Set brown-out detection
board_hardware.eesave = yes          ; Preserve EEPROM when uploading using programmer
upload_protocol = usbasp             ; Use the USBasp as programmer
upload_flags =                       ; Select USB as upload port and divide the SPI clock by 8
  -PUSB
  -B8
```

trace
```
Processing Upload_UART (platform: atmelavr; framework: arduino; board: ATmega1284P)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega1284P.html
PLATFORM: Atmel AVR (4.0.0) > ATmega1284P
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 128KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr-mightycore @ 2.1.3       
 - tool-avrdude @ 1.60300.200527 (6.3.0)
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)    
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 14 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio\build\Upload_UART\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.1% (used 9 bytes from 16384 bytes)
Flash: [          ]   0.9% (used 1160 bytes from 130048 bytes)
Configuring upload protocol...
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Auto-detected: COM5
Uploading .pio\build\Upload_UART\firmware.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: reading input file ".pio\build\Upload_UART\firmware.hex"
avrdude: writing flash (1160 bytes):

Writing | ################################################## | 100% 0.24s

avrdude: 1160 bytes of flash written
avrdude: verifying flash memory against .pio\build\Upload_UART\firmware.hex:
avrdude: load data flash data from input file .pio\build\Upload_UART\firmware.hex:
avrdude: input file .pio\build\Upload_UART\firmware.hex contains 1160 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.26s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0002
         0x53 != 0x80
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

*** [upload] Error 1
```

MCUdude, 2022-09-14:
@Sercurio can you post the PlatformIO output when running the following command?
The -D flag might be missing, but since PlatformIO doesn't reveal the Avrdude command and its flag when not in verbose mode, there's no way to tell.

`pio run -t upload -e Upload_UART -v`

Sercurio, 2022-09-15:
```
Processing Upload_UART (upload_protocol: arduino; board_upload.speed: 115200; lib_deps: adafruit/Adafruit GFX Library@^1.10.15, adafruit/Adafruit SSD1306@^2.5.3; platform: atmelavr; framework: arduino; board: ATmega1284P; board_build.f_cpu: 16000000L; board_build.variant: standard; build_unflags: -flto; build_flags: ; monitor_port: None; monitor_speed: 9600)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega1284P.html
PLATFORM: Atmel AVR (4.0.0) > ATmega1284P
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 128KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr-mightycore @ 2.1.3
 - tool-avrdude @ 1.60300.200527 (6.3.0)
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 17 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit GFX Library @ 1.11.3 (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\.pio\libdeps\Upload_UART\Adafruit GFX Library)
|   |-- Adafruit BusIO @ 1.12.0 (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\.pio\libdeps\Upload_UART\Adafruit BusIO)
|   |   |-- Wire @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\Wire)
|   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\SPI)
|   |-- Wire @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\Wire)
|   |-- SPI @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\SPI)
|-- Adafruit SSD1306 @ 2.5.6 (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\.pio\libdeps\Upload_UART\Adafruit SSD1306)
|   |-- Adafruit GFX Library @ 1.11.3 (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\.pio\libdeps\Upload_UART\Adafruit GFX Library)
|   |   |-- Adafruit BusIO @ 1.12.0 (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\.pio\libdeps\Upload_UART\Adafruit BusIO)
|   |   |   |-- Wire @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\Wire)
|   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\SPI)
|   |   |-- Wire @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\Wire)
|   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\SPI)
|   |-- Adafruit BusIO @ 1.12.0 (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\.pio\libdeps\Upload_UART\Adafruit BusIO)
|   |   |-- Wire @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\Wire)
|   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\SPI)
|   |-- Wire @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\Wire)
|   |-- SPI @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\SPI)
|-- Bb (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\lib\Bb)
|   |-- EEPROM @ 2.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\EEPROM)
|   |-- Adafruit SSD1306 @ 2.5.6 (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\.pio\libdeps\Upload_UART\Adafruit SSD1306)
|   |   |-- Adafruit GFX Library @ 1.11.3 (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\.pio\libdeps\Upload_UART\Adafruit GFX Library)
|   |   |   |-- Adafruit BusIO @ 1.12.0 (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\.pio\libdeps\Upload_UART\Adafruit BusIO)
|   |   |   |   |-- Wire @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\Wire)
|   |   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\SPI)
|   |   |   |-- Wire @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\Wire)
|   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\SPI)
|   |   |-- Adafruit BusIO @ 1.12.0 (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\.pio\libdeps\Upload_UART\Adafruit BusIO)
|   |   |   |-- Wire @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\Wire)
|   |   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\SPI)
|   |   |-- Wire @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\Wire)
|   |   |-- SPI @ 1.0 (License: Unknown, Path: C:\.platformio\packages\framework-arduino-avr-mightycore\libraries\SPI)
|   |-- Adafruit GFX Library @ 1.11.3 (License: Unknown, Path: C:\Users\lpenalva\Documents\Projects\bytebeatOLED\.pio\libdeps\Upload_UART\Adafruit GFX Library)
[... cut ...]
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

*** [upload] Error 1
=========================================================================== [FAILED] Took 20.27 seconds ===========================================================================
Environment       Status    Duration
----------------  --------  ------------
Upload_UART       FAILED    00:00:20.266
Upload_ISP        IGNORED
fuses_bootloader  IGNORED
```

MCUdude, 2023-11-30:
Issue outdated now that Optiboot has been replaced by Urboot. Try installing the latest MightyCore package and re-burn the bootloader