MCUdude/MightyCore #199: Atmel Studio Failed compiling in expansion of macro round

muluyilmaz, 2020-12-07:
After updating to 2.0.7 I got this error  at ATmel Studio

arduino.h: 128:89: error: expected ';' before '}' token (long)_x - 0.5 })

MCUdude, 2020-12-07:
Thanks for reporting. Are you using the `round()` function somewhere in your code?

tgidave, 2020-12-13:
The same message is being generated during expansion of the "constrain" macro.  The DallasTemperature library uses this macro.

Dave

closed 2020-12-14 by MCUdude

JensGrabner, 2020-12-14:
The last Version that work for me is Version 2.0.6
https://github.com/JensGrabner/snc98_Slash-Number-Calculator/tree/master/Software/Arduino/Projekte/snc98

MCUdude, 2020-12-14:
> The last Version that work for me is Version 2.0.6

What exactly is it that doesn't work since 2.0.6? Have you tried 2.0.8?

JensGrabner, 2020-12-14:
28-11-2020  --  avrdude: 106796 bytes of flash written    // Version 2.0.6
05-12-2020  --  avrdude: 105644 bytes of flash written    // Version 2.0.7 defect
Version 2.0.8 have the same effect then Version 2.0.7

I use a lot of global variable  --  my project ist not working - i cant say what it is?  --  Start is o.k. but when i calculate 1/3 came a difuse display. other compiler  -  i use the Arduino IDE 1.8.13

MCUdude, 2020-12-14:
You have all the changes here:
https://github.com/MCUdude/MightyCore/releases

Just look through the commits and revert them on your computer to find out what causes the issue. It might be the new `abs()` macro.

MCUdude, 2020-12-14:
You can try to replace the abs function with the pld implementation:

`#define old_abs(x) ((x)>0?(x):-(x))`

JensGrabner, 2020-12-14:
the different in compiling:
=====
compare from (<)C:\Users\jens\Documents\2_0_8.txt (86214 Byte)
 to (>)C:\Users\jens\Documents\2_0_6.txt (90024 Byte)

113,115c113,115
\\MCUdude_corefiles\\CDC.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\CDC.cpp.o"
\\MCUdude_corefiles\\HardwareSerial.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\HardwareSerial.cpp.o"
\\MCUdude_corefiles\\HardwareSerial1.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\HardwareSerial1.cpp.o"

\\MCUdude_corefiles\\HardwareSerial1.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\HardwareSerial1.cpp.o"
\\MCUdude_corefiles\\CDC.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\CDC.cpp.o"
\\MCUdude_corefiles\\HardwareSerial.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\HardwareSerial.cpp.o"
171c171
< Sketch uses 105956 bytes (81%) of program storage space. Maximum is 130048 bytes.
---
> Sketch uses 107108 bytes (82%) of program storage space. Maximum is 130048 bytes.
======

not the same order from
- CDC.cpp
- HardwareSerial.cpp
- HardwareSerial1.cpp

JensGrabner, 2020-12-14:
Vergleichen von (<)C:\Users\jens\Documents\2_0_6.txt (90004 Byte)
 mit (>)C:\Users\jens\Documents\2_0_8.txt (86194 Byte)

67,73c67,73
< "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284 -DF_CPU=12000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\jens\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.x\\cores\\MCUdude_corefiles" "-IC:\\Users\\jens\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.x\\variants\\standard" "-IC:\\Program Files (x86)\\Arduino\\libraries\\OneWire" "-IC:\\Program Files (x86)\\Arduino\\libraries\\BitBool" "-IC:\\Program Files (x86)\\Arduino\\libraries\\RingBufCPP" "-IC:\\Program Files (x86)\\Arduino\\libraries\\int96" "-IC:\\Program Files (x86)\\Arduino\\libraries\\itoa_ljust" "-IC:\\Program Files (x86)\\Arduino\\libraries\\TimerOne" "-IC:\\Program Files (x86)\\Arduino\\libraries\\OneWireHub\\src" "C:\\Program Files (x86)\\Arduino\\libraries\\BitBool\\BitBool.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\libraries\\BitBool\\BitBool.cpp.o"
< Compiling library "RingBufCPP"
< Compiling library "int96"
< "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284 -DF_CPU=12000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\jens\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.x\\cores\\MCUdude_corefiles" "-IC:\\Users\\jens\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.x\\variants\\standard" "-IC:\\Program Files (x86)\\Arduino\\libraries\\OneWire" "-IC:\\Program Files (x86)\\Arduino\\libraries\\BitBool" "-IC:\\Program Files (x86)\\Arduino\\libraries\\RingBufCPP" "-IC:\\Program Files (x86)\\Arduino\\libraries\\int96" "-IC:\\Program Files (x86)\\Arduino\\libraries\\itoa_ljust" "-IC:\\Program Files (x86)\\Arduino\\libraries\\TimerOne" "-IC:\\Program Files (x86)\\Arduino\\libraries\\OneWireHub\\src" "C:\\Program Files (x86)\\Arduino\\libraries\\int96\\int96.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\libraries\\int96\\int96.cpp.o"
< "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284 -DF_CPU=12000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\jens\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.x\\cores\\MCUdude_corefiles" "-IC:\\Users\\jens\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.x\\variants\\standard" "-IC:\\Program Files (x86)\\Arduino\\libraries\\OneWire" "-IC:\\Program Files (x86)\\Arduino\\libraries\\BitBool" "-IC:\\Program Files (x86)\\Arduino\\libraries\\RingBufCPP" "-IC:\\Program Files (x86)\\Arduino\\libraries\\int96" "-IC:\\Program Files (x86)\\Arduino\\libraries\\itoa_ljust" "-IC:\\Program Files (x86)\\Arduino\\libraries\\TimerOne" "-IC:\\Program Files (x86)\\Arduino\\libraries\\OneWireHub\\src" "C:\\Program Files (x86)\\Arduino\\libraries\\int96\\StdAfx.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\libraries\\int96\\StdAfx.cpp.o"
< Compiling library "itoa_ljust"
< "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284 -DF_CPU=12000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\jens\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.x\\cores\\MCUdude_corefiles" "-IC:\\Users\\jens\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.x\\variants\\standard" "-IC:\\Program Files (x86)\\Arduino\\libraries\\OneWire" "-IC:\\Program Files (x86)\\Arduino\\libraries\\BitBool" "-IC:\\Program Files (x86)\\Arduino\\libraries\\RingBufCPP" "-IC:\\Program Files (x86)\\Arduino\\libraries\\int96" "-IC:\\Program Files (x86)\\Arduino\\libraries\\itoa_ljust" "-IC:\\Program Files (x86)\\Arduino\\libraries\\TimerOne" "-IC:\\Program Files (x86)\\Arduino\\libraries\\OneWireHub\\src" "C:\\Program Files (x86)\\Arduino\\libraries\\itoa_ljust\\itoa_ljust.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\libraries\\itoa_ljust\\itoa_ljust.cpp.o"
---
> Using previously compiled file: C:\Users\jens\AppData\Local\Temp\arduino_build_80951\libraries\BitBool\BitBool.cpp.o
> Compiling library "RingBufCPP"
> Compiling library "int96"
> Using previously compiled file: C:\Users\jens\AppData\Local\Temp\arduino_build_80951\libraries\int96\int96.cpp.o
> Using previously compiled file: C:\Users\jens\AppData\Local\Temp\arduino_build_80951\libraries\int96\StdAfx.cpp.o
> Compiling library "itoa_ljust"
> Using previously compiled file: C:\Users\jens\AppData\Local\Temp\arduino_build_80951\libraries\itoa_ljust\itoa_ljust.cpp.o
98,99c98,99
< "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega1284 -DF_CPU=12000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\jens\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.x\\cores\\MCUdude_corefiles" "-IC:\\Users\\jens\\AppData\\Local\\ [... cut ...]
> cmd /C "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951/snc98.ino.elf" > "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951/snc98.ino_atmega1284_12000000L.lst"
161c161
< Sketch uses 107108 bytes (82%) of program storage space. Maximum is 130048 bytes.
---
> Sketch uses 105956 bytes (81%) of program storage space. Maximum is 130048 bytes.

JensGrabner, 2020-12-14:
the main difference in compiling:

=====
compare from (<)C:\Users\jens\Documents\2_0_8.txt (46684 Byte)
 to (>)C:\Users\jens\Documents\2_0_6.txt (47426 Byte)

98,99c98,99
< "\\wiring_analog.c" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\wiring_analog.c.o"
< "\\wiring.c" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\wiring.c.o"
---
> "\\wiring.c" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\wiring.c.o"
> "\\wiring_analog.c" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\wiring_analog.c.o"
103,105c103,105
< "\\CDC.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\CDC.cpp.o"
< "\\HardwareSerial.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\HardwareSerial.cpp.o"
< "\\HardwareSerial1.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\HardwareSerial1.cpp.o"
---
> "\\HardwareSerial1.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\HardwareSerial1.cpp.o"
> "\\CDC.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\CDC.cpp.o"
> "\\HardwareSerial.cpp" -o "C:\\Users\\jens\\AppData\\Local\\Temp\\arduino_build_80951\\core\\HardwareSerial.cpp.o"
=====

commit f72f2c7d (2020-12-14): Add 2.0.8 to boards manager
Closes #199