MCUdude/MightyCore #202: Unable to toggle Led on Port A, C,D except Port B

dnvasta, 2020-12-30:
Running a basic sketch, i'm not able to toggle a LED on Port A,C,D but works fine on Port B
Any reasons ?

Fuses : (E:FF, H:D4, L:A4)
8Mhz, internal clock

`
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  //pinMode(PB0,OUTPUT);
  
  portMode(2, OUTPUT);  //PortC
}

void loop() {
  // put your main code here, to run repeatedly:
  //digitalWrite(PB0,HIGH);
  portWrite(2,HIGH);
  delay(3000);
  //digitalWrite(PB0,LOW);
  portWrite(2,LOW);
  delay(3000);
}
`

MCUdude, 2020-12-30:
Are you trying to flash all pins on the port? If so, it's not going to work the way you're trying to do it.

```
portWrite(2,HIGH);
```
Will only set the first pin on the port (PA0, PB0, PC0, PD0). If you want to toggle all pins of the port, use

```
portWrite(2, 0xff);

dnvasta, 2020-12-31:
Thanks @MCUdude it works!

closed 2020-12-31 by dnvasta