MCUdude/MightyCore #203: Bootloader command line contains placeholder variables

ianliu, 2021-01-06:
I'm trying to burn the MightyCore bootloader with Arduino IDE, but I'm getting the following message:

```
Arduino: 1.8.13 (Linux), Board: "ATmega32, Yes (UART0), EEPROM retained, Standard pinout, BOD 2.7V, LTO disabled, External 16 MHz"

{runtime.tools.avrdude.path}/bin/avrdude -C/home/ian/.arduino15/packages/MightyCore/hardware/avr/2.0.8/avrdude.conf -v -patmega32 -cstk500v1 -P/dev/ttyACM0 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0b11000110:m -Ulfuse:w:0b10111111:m 
java.io.IOException: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:621)
	at java.lang.Runtime.exec(Runtime.java:486)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:372)
	at processing.app.Editor.lambda1(Editor.java:2379)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 8 more
Error while burning bootloader.
```

It seems some placeholder variables aren't being replaced.

MCUdude, 2021-01-06:
What happens if you install the latest Arduino Atmel AVR package from the boards manager (1.8.3)?

ianliu, 2021-01-06:
That fixed the arvdude command path, but the `-Uefuse:w:{bootloader.extended_fuses}:m` section was left as-is. Nonetheless, it worked!

MCUdude, 2021-01-06:
Great to hear it worked!

`-Uefuse:w:{bootloader.extended_fuses}:m` is supposed to be there. The avrdude.conf file is slightly modified so that AVRs without efuse (like the ATmega32) will ignore this flag. This is done to simplify the platform.txt file.

closed 2021-01-06 by MCUdude