MCUdude/MightyCore #204: Example compile error

feco9308, 2021-01-18:
Hi!

I don't know what I messed up, I successfully installed the plugin. Bootloader was also successfully updated.
And when we say the Blink is sketched I want to compile the following error message:

`C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\feco9\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\feco9\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\feco9\Documents\Arduino\libraries -fqbn=MightyCore:avr:1284:bootloader=uart0,pinout=standard,eeprom=keep,variant=modelNonP,BOD=2v7,LTO=Os,clock=16MHz_external -ide-version=10813 -build-path C:\Users\feco9\AppData\Local\Temp\arduino_build_811269 -warnings=none -build-cache C:\Users\feco9\AppData\Local\Temp\arduino_cache_627430 -prefs=build.warn_data_percentage=75 -verbose C:\Users\feco9\Documents\ATmega128\Blink\Blink.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\feco9\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\feco9\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\feco9\Documents\Arduino\libraries -fqbn=MightyCore:avr:1284:bootloader=uart0,pinout=standard,eeprom=keep,variant=modelNonP,BOD=2v7,LTO=Os,clock=16MHz_external -ide-version=10813 -build-path C:\Users\feco9\AppData\Local\Temp\arduino_build_811269 -warnings=none -build-cache C:\Users\feco9\AppData\Local\Temp\arduino_cache_627430 -prefs=build.warn_data_percentage=75 -verbose C:\Users\feco9\Documents\ATmega128\Blink\Blink.ino
Using board '1284' from platform in folder: C:\Users\feco9\Documents\ArduinoData\packages\MightyCore\hardware\avr\2.0.8
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\feco9\Documents\ArduinoData\packages\MightyCore\hardware\avr\2.0.8
Detecting libraries used...
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles" "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\variants\\standard" "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269\\sketch\\Blink.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles" "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\variants\\standard" "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269\\sketch\\Blink.ino.cpp" -o "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles" "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\variants\\standard" "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269\\sketch\\Blink.ino.cpp" -o "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269\\sketch\\Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles" "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\variants\\standard" "C:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles\\wiring_pulse.S" -o "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269\\core\\wiring_pulse.S.o"
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles" "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\variants\\standard" "C:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles\\wiring.c" -o "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269\\core\\wiring.c.o"
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles" "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\variants\\standard" "C:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles\\wiring_digital.c" -o "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269\\core\\wiring_digital.c.o"
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles" "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\variants\\standard" "C:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles\\wiring_analog.c" -o "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269\\core\\wiring_analog.c.o"
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega1284 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR "-IC:\\Users\\feco9\\Documents\\ArduinoData\\packages\\MightyCore\\hardware\\avr\\2.0.8\\cores\\MCUdude_corefiles" "-IC:\\Users\\feco9 [... cut ...]
cmd /C "C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269/Blink.ino.elf" > "C:\\Users\\feco9\\AppData\\Local\\Temp\\arduino_build_811269/Blink.ino_atmega1284_16000000L.lst"
A hozzáférés megtagadva.
exit status 1
Error compiling for board ATmega1284.
`

Anyone ready for an idea?
(Sorry for the language errors.)

MCUdude, 2021-01-18:
Hi! Have you installed Arduino IDE from the Windows store? If yes, uninstall this version and download IDE 1.8.13 from https://arduino.cc .
According to the error message, there is a script that doesn't have permission to run.

feco9308, 2021-01-18:
Thanks for the quick help! The problem is solved

MCUdude, 2021-01-18:
Great to hear it worked out for you!

closed 2021-01-18 by MCUdude