MCUdude/MightyCore #206: ATmega32 low power with WDT wakeup help needed

sensorhost, 2021-02-11:
I have code from an ATmega328P (using MiniCore) and it works well, but I am setting registers directly.

I have now moved to using an ATmega32 and the code doesn't work and I have searched and searched this repository for any clues to help me without success.

Basically, I want to put the device into `SLEEP_MODE_PWR_DOWN` for 250mS, then wake up using the WDT and continue where it left off, I don't want the device to reset. Normally when I wake up, I re-enable the WDT to reset the device if the code gets locked.

I have an ISR as follows, which the compiler complains about `warning: 'WDT_vect' appears to be a misspelled signal handler, missing __vector prefix [-Wmisspelled-isr]`
```
ISR (WDT_vect) 
{
  wdt_reset();
  wdt_disable();  // disable watchdog
}
```

I would appreciate if someone could provide a simple example that could help me get started.
Thanks.

MCUdude, 2021-02-11:
The ATmega32 doesn't support WDT interrupts. AFAIK you can only use the WDT to reset the microcontroller if it hangs. If you need  WDT interrupts you'll have to use a more modern microcontroller, such as ATmega324.

sensorhost, 2021-02-11:
Hi @MCUdude, thanks for such a quick response.

Well that kind of sucks, but it is my fault for not digging into the data sheet. It was the only MCU the PCB fab had available (previously tried the 644 and it works well).

Closing as a non-issue.

closed 2021-02-11 by sensorhost