MCUdude/MightyCore #209: Can't compile code when SPI1 is used in Arduino IDE

shakil-anwar, 2021-03-14:
I get error: 'SPCR1' was not declared in this scope

C:\Users\Asus\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.0\libraries\SPI1\src/SPI1.h:292:31: note: suggested alternative: 'SPCR'

     if (bitOrder == LSBFIRST) SPCR1 |= _BV(DORD);

                               ^~~~~
                               SPCR

I am getting many similar errors , while compiling with SPI1.begin() and SPI1.transfer(..,..);

MCUdude, 2021-03-14:
Which microcontroller are you compiling for?  ATmega1284P?

shakil-anwar, 2021-03-14:
yes I was trying to compile for atmega1284p

MCUdude, 2021-03-14:
The 1284P only has one SPI peripheral. SPI1 is only for the ATmega324PB.

closed 2021-03-14 by MCUdude

tomaskovacik, 2021-03-14:
seems like there is a typo on the product page 
https://www.microchip.com/wwwproducts/en/ATmega1284P
"Digital Communication Peripherals 2-UART, 3-SPI, 1-I2C"

the datasheet talks about only one SPI interface

but there is the possibility to create a second SPI master using USART, check datasheet 

 "The Universal Synchronous and Asynchronous serial Receiver and Transmitter (USART) can be set to a master SPI compliant mode of operation."