MCUdude/MightyCore #216: Zła kompilacja na Atmega 644PA

Gerston, 2021-05-10:
Jestem nowicjuszem w programowaniu procesorów, napotkałem taki problem, zainstalowałem bibliotekę MightyCore do arduino IDE na procesory Atmega, między innymi Atmega 644, który stosuję w moim projekcie. Zakupiłem procesor Atmega 644PA, w/g specyfikacji może być taktowany zegarem 20MHz, zastosowałem taki kwarc. Od razu pojawiły się problemy przy kompilacji kodu, który działał be zarzutów na arduino UNO 16MHz! 
Niestety arduino UNO ma za mało pamięci na mój projekt, więc postanowiłem zastosować inny procesor, który sprosta moim oczekiwaniom!
Po pierwsze nie mogłem wgrać bootloadera, ale pomijam ten problem, to nie jest mi niezbędne do życia!!!
Po drugie, przy kompilacji na 20MHz program po uruchomieniu się od razu wieszał!
Oki, stwierdziłem, że skompiluję go na zegarze wewnętrznym 1MHz, program działał dobrze, ale nie odczytywał danych z czujników temperatury i wilgotności z DHT11. Skompilowałem program z zegarem wewnętrznym 8MHz, na wyświetlaczu pojawiły się dziwne znaczki, zamiast polskich znaków, które zdefiniowałem w PROGMEM jakieś inne i w ogóle działy się  dziwne rzeczy na wyświetlaczu, czujnik temperatury nie był dalej odczytywany! Próbowałem z różnymi konfiguracjami sprzętu, efekt był ten sam!!! Dopiero jak zamieniłem kwarc na 16MHz problem ustąpił!!! Chyba nie powinno tak być, że procesory jeśli są taktowane zegarem 16MHz to wszystko działa dobrze, a jeśli innym niż założenie przewiduje to nie działają?

MCUdude, 2021-05-11:
Hi!
Sorry, I don't understand a word of Polish, so I had to put your text through Google Translate. I only understand English and Norwegian.

You have described the problem you're seeing, but not provided any information so I can give you any meaningful help.

You don't have to use a bootloader at all. And I believe you've been able to set the correct fuses and upload your program with an ISP programmer? If you don't set the fuses the microcontroller will not run at the desired speed. Select `Bootloader: No bootloader` in the Tools menu, and then run `Burn Bootloader`. This is most likely the root cause of the problems you're seeing.

As a start, I would highly recommend you to use "regular" English letters on your screen for now. Special unicode characters usually take up more than 8 bit, which may cause a problem when you have a fixed allocated flash space.

Gerston, 2021-05-11:
Hi!
Thank you very much for your quick reply :)

You were right, the problem was that the fuses were blown properly.
I tested it on different clocks and everything works fine!

Thanks again for your help :)

closed 2021-05-11 by Gerston