MCUdude/MightyCore #217: Can't get serial monitor in atmega32 with Arduino Uno as ISP

muhit313, 2021-05-19:
I was trying to burn atmega32 with Arduino Uno as ISP using [this tutorial](https://www.instructables.com/Burning-Arduino-Bootloader-to-Atmega-32A-PU-Using-/) which using **MightyCore** library and successfully do this.

Now I can upload any code in my atmega32 using Arduino and all work fine. But the problem  happened when I was  trying to upload serial monitor code. The code successfully upload but i cant see anything on serial monitor.

**N.B:** I also trying by connecting Rx on my Arduino with Tx of atmega32 and  Tx on my Arduino with Rx of atmega32.
but nothing work

MCUdude, 2021-05-19:
Hi!

> Now I can upload any code in my atmega32 using Arduino and all work fine.

Are you able to upload to the ATmega32 using a USB to serial adapter? Or are you still using ISP?

RX and TX do work, so it's either something wrong with your code or your hardware.

muhit313, 2021-05-19:
> Are you able to upload to the ATmega32 using a USB to serial adapter? Or are you still using ISP?
No. I don't have USB to serial adapter. I'm still using ISP.
I have a Usb to ttl. Can I use **Usb to ttl** instated of USB to serial adapter? 
Sorry for that, I don't no what is the difference between them.
> RX and TX do work, so it's either something wrong with your code or your hardware.
I was trying your **MightyCore** **SoftwareSerialExample** code.
Basically this code:
`#include <SoftwareSerial.h>

SoftwareSerial mySerial(10, 11); // RX, TX

void setup() 
{
  // Open serial communications and wait for port to open:
  Serial.begin(57600);
  while (!Serial); // wait for serial port to connect. Needed for native USB port only 

  Serial.println("Goodnight moon!");

  // Set the data rate for the SoftwareSerial port
  mySerial.begin(4800);
  mySerial.println("Hello, world?");
}

void loop() 
{ 
  if (mySerial.available()) 
  {
    Serial.write(mySerial.read());
  }
   
  if (Serial.available()) 
  {
    mySerial.write(Serial.read());
  }
}`

muhit313, 2021-05-19:
No. I don't have USB to serial adapter. I'm still using ISP.
I have a Usb to ttl. Can I use Usb to ttl instated of USB to serial adapter?
Sorry for that, I don't no what is the difference between them.
I was trying your MightyCore SoftwareSerialExample code.
Basically this code:
```
#include <SoftwareSerial.h>

SoftwareSerial mySerial(10, 11); // RX, TX

void setup()
{
// Open serial communications and wait for port to open:
Serial.begin(57600);
while (!Serial); // wait for serial port to connect. Needed for native USB port only

Serial.println("Goodnight moon!");

// Set the data rate for the SoftwareSerial port
mySerial.begin(4800);
mySerial.println("Hello, world?");
}

void loop()
{
if (mySerial.available())
{
Serial.write(mySerial.read());
}

if (Serial.available())
{
mySerial.write(Serial.read());
}
}
```

muhit313, 2021-05-20:
Finally I can see my serial monitor with Usb to ttl.
May be Arduino as ISP is not working at all with atmega32.

closed 2021-05-20 by muhit313

tomaskovacik, 2021-05-20:
arduino as ISP is programmer not usb2serial converter, if you like to use arduino board as usb2serial converter, you must connect reset ping of this arduino to ground(it will not work at all)  and connect rx and txt pins plus ground to your tager (do not foger swap tx and rx)