MCUdude/MightyCore #219: Arduino IDE 2.0-beta.7 does not have baud rates beyond 115200

moulisankaran, 2021-06-04:
The previous versions had support for beyond 115200, till 2000000 baud rates.

Moreover, while trying to set the drop down menu on the right hand side, the pop error messages appearing on the bottom right, hides the baud rate drop-down option too.

Please fix these issues.

[screeshot.pdf](https://github.com/MCUdude/MightyCore/files/6596797/screeshot.pdf)

per1234, 2021-06-04:
@moulisankaran that has nothing to do with MightyCore. It is a deficiency in Arduino IDE 2.x itself, which is already being tracked here:
https://github.com/arduino/arduino-ide/issues/111

For now, you'll need to use the classic Arduino IDE or a standalone serial terminal if you need baud rates other than those provided by the Serial Monitor.

moulisankaran, 2021-06-04:
Sure, thanks for your response.

closed 2021-06-04 by moulisankaran