MCUdude/MightyCore #220: min/max bug

leetut, 2021-07-09:
my code won't compile using mightycore,
I was told its because of a bug in the MightyCore? 
min and max shouldn't be non-standard statement-expressions, but function templates:
`
  template<class T, class L> 
   auto min(const T& a, const L& b) -> decltype((b < a) ? b : a)
   {
    return (b < a) ? b : a;
  }

          
  template<class T, class L> 
   auto max(const T& a, const L& b) -> decltype((b < a) ? b : a)
   {
    return (a < b) ? b : a;
  }
#else
#ifndef min
#define min(a,b) \
    ({ __typeof__ (a) _a = (a); \
        __typeof__ (b) _b = (b); \
      _a < _b ? _a : _b; })
#endif
#ifndef max`

my compile error:
`/Users/lee/Library/Arduino15/packages/MightyCore/hardware/avr/2.1.3/cores/MCUdude_corefiles/Arduino.h:126:22: error: statement-expressions are not allowed outside functions nor in template-argument lists
 #define min(a,b)     ({ typeof (a) _a = (a); typeof (b) _b = (b); _a < _b ? _a : _b; })
                      ^
/Users/lee/Desktop/fvduino Atmega1284 Breadboard/fvduino/fvduino.ino:240:23: note: in expansion of macro 'min'
 const uint8_t NALGO = min(99, sizeof(algodata) / sizeof(AlgoDatum));
                       ^~~
Using library EEPROM at version 2.0 in folder: /Users/lee/Library/Arduino15/packages/MightyCore/hardware/avr/2.1.3/libraries/EEPROM 
Using library ButtonEvents-master at version 1.0.1 in folder: /Users/lee/Documents/Arduino/libraries/ButtonEvents-master 
Using library Bounce2 at version 2.56 in folder: /Users/lee/Documents/Arduino/libraries/Bounce2 
Using library Adafruit_NeoPixel at version 1.7.0 in folder: /Users/lee/Documents/Arduino/libraries/Adafruit_NeoPixel 
exit status 1
Error compiling for board ATmega1284.`

anyway to fix this?

MCUdude, 2021-07-11:
You can use the original Arduino implementation:

```c
#undef min
#define min(a,b) ((a)<(b)?(a):(b))
```

leetut, 2021-07-12:
Thank you dude

closed 2022-04-24 by MCUdude