MCUdude/MightyCore #221: atmega32-mighty-core-ps2-keyboard

madhavanatmel, 2021-08-25:
hi, i am trying to connect ps2 keyboard in atmega32 and using pin no . 16&17 as Arduino pin 10&11 for ps2 keyboard Data pin & IRQ Pin . But i couldn't get any keycode in the serial monitor . at the beginning KEYBOARD test message is from m32 after that nothing is coming .
[image]
the same keyboard and ps2 keyboard library working well with uno using Arduino pin as 2& 3

madhavanatmel, 2021-08-29:
Good that now working ..
#elif defined(__AVR_ATmega644P__) || defined(__AVR_ATmega32__) // Sanguino.
standard pin layout .

MCUdude, 2021-08-29:
Great to hear!

closed 2021-08-29 by MCUdude