MCUdude/MightyCore #222: Serial.printf not working with Platformio

toogooda, 2021-09-09:
I love your core with Arduino but I can't work out how to use it with platformio, when selecting boards nothing shows up for Mightycore or MCUdude. Your platformio.ini example doesn't seem to reference mightycore at all so it was hard to see what to add to be using your core.
I have tried to follow your platformio guide to get the printf working but the suggested build flag does not work.

```
[env:644pa8m]
platform = atmelavr
board = 644pa8m
framework = arduino
lib_deps = 
	chris--a/Keypad@^3.1.1
build_flags = -lprintf_flt
```

printf is undefined...

MCUdude, 2021-09-09:
Hi! `board = 644pa8m` is not a valid MightyCore board type, So you're probably using the old Sanguino core without knowing.

If you look at the [MightyCore PlatformIO README](https://github.com/MCUdude/MightyCore/blob/master/PlatformIO.md#board), you can see that you'll have to use `board = ATmega644P` instead. Then you'll have to specify your clock frequency with `board_build.f_cpu`.

toogooda, 2021-09-09:
Thank you I have it working now and the template was very helpful now that I worked out I can just change the default at the top change between USB/ISP/Bootloader.
I don't see anything listed under MCUDUDE or Mightycore when I use the create project from Platformio home the ones listed in your readme seem to be (Microchip)...

So is the correct procedure to use Mightycore in Platformio to chose any board then overwrite the ini with the template?

Thanks again for your help, your cores are legendary!

closed 2021-09-09 by toogooda