MCUdude/MightyCore #223: ATMEGA324PB Pinout Issues with Ethernet 5500

JoaquimBoavida, 2021-09-16:
Hi,
I have a board (my own design) with an ATmega324PB and a WIZ850io module. The board is pretty normal design with the MISO, MOSI SCK and SS pins of the  Ethernet module connected to PB5, PB6, PB7 and PB4 pins of the Mega.
I also use the UART1 (RX1, TX1) on pins PD2 and PD3.

When I use the "standard" pinout the serial port works perfectly but the WIZ850io (that uses a W5500 chip) is not detected.
When I use the "Bobuino" pinout, the ethenet module is detected, and "pings" fine but the serial port does no work properly and I get garbage.

What i'm doing wrong?

Thanks

Joaquim

MCUdude, 2021-09-16:
What ethernet library are you using? Please provide a link to the repo. And do you run the 324PB from its internal oscillator, or from an external crystal?

I happened to have this exact module in my drawer. Can you provide a bare-bone testsketch I can try?

JoaquimBoavida, 2021-09-16:
H, Please find the files attached.
The Library is the built in (picture attached).
I don't know what happend but suddenly started to work with bobuino pinout
Please check. Note that Serial1 is RS485.
The problem happens more often on board power on.
Thanks

Joaquim
[image]
[EtherTest.zip](https://github.com/MCUdude/MightyCore/files/7180716/EtherTest.zip)

MCUdude, 2021-09-16:
Ok, I've done a little research and figured out why you're not getting it to work properly.

I'm not sure which pin you use for CS/SS, but you'll have to run `Ethernet.init(4 /* some pin */);` _before_ `Ethernet.init()`

This code works for me. I'm using SPI0 on the ATmega324PB (standard pinout), and I'm using digital pin 4/PB4 as the SS/CS pin. 

```c++
#include <SPI.h>
#include <Ethernet.h>
#include <Wire.h>
#define MAC_ADDRESS 0x50

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
byte ipa[] = { 0, 0, 0, 0 }; //adress
byte ipd[] = { 0, 0, 0, 0 }; //dns
byte ipg[] = { 0, 0, 0, 0 }; //Gateway
byte ips[] = { 0, 0, 0, 0 }; //Subnet Mask

int RS485DE = 28;
//int ETHRST = 29;

// Enter a MAC address and IP address for your controller below.
// The IP address will be dependent on your local network.
// gateway and subnet are optional:

IPAddress ip(192, 168, 1, 177);
IPAddress myDns(192, 168, 1, 1);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 0, 0);

// telnet defaults to port 23

EthernetServer server(5000);

bool alreadyConnected = false; // whether or not the client was connected previously

void setup() {

  pinMode(RS485DE, OUTPUT);
  digitalWrite(RS485DE, HIGH);
  //pinMode(ETHRST, OUTPUT);
  //digitalWrite(ETHRST, HIGH);
  
  Wire.begin();         // Join i2c bus (I2C address is optional for the master)
  Serial1.begin(115200);

/*  mac[0] = readRegister(0xFA);
  mac[1] = readRegister(0xFB);
  mac[2] = readRegister(0xFC);
  mac[3] = readRegister(0xFD);
  mac[4] = readRegister(0xFE);
  mac[5] = readRegister(0xFF);
 
  ipa[0] = readRegister(0x01);
  ipa[1] = readRegister(0x02);
  ipa[2] = readRegister(0x03);
  ipa[4] = readRegister(0x04);
  
  ipd[0] = readRegister(0x05);
  ipd[1] = readRegister(0x06);
  ipd[2] = readRegister(0x07);
  ipd[4] = readRegister(0x08);

  ipg[0] = readRegister(0x09);
  ipg[1] = readRegister(0x0A);
  ipg[2] = readRegister(0x0B);
  ipg[4] = readRegister(0x0C);

  ips[0] = readRegister(0x0D);
  ips[1] = readRegister(0x0E);
  ips[2] = readRegister(0x0F);
  ips[4] = readRegister(0x10);
*/

   Ethernet.init(4);  // Most Arduino shields
  Ethernet.begin(mac, ip, myDns, gateway, subnet);

  // You can use Ethernet.init(pin) to configure the CS pin
   // Most Arduino shields
  // Check for Ethernet hardware present

  if (Ethernet.hardwareStatus() == EthernetNoHardware) {
    Serial1.println("Ethernet shield was not found.");
  }
  else if (Ethernet.hardwareStatus() == EthernetW5100) {
    Serial1.println("W5100 Ethernet controller detected.");
  }
  else if (Ethernet.hardwareStatus() == EthernetW5200) {
    Serial1.println("W5200 Ethernet controller detected.");
  }
  else if (Ethernet.hardwareStatus() == EthernetW5500) {
    Serial1.println("W5500 Ethernet controller detected.");
  }

//  if (Ethernet.hardwareStatus() == EthernetNoHardware) {
//    Serial1.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
 //   while (true) {
  //    delay(1); // do nothing, no point running without Ethernet hardware
  //  }
 // }

  if (Ethernet.linkStatus() == LinkOFF) {
    Serial1.println("Ethernet cable is not connected.");
   }
  // start listening for clients
  server.begin();
  Serial1.print("Chat server address:");
  Serial1.println(Ethernet.localIP());
}

void loop() {
  // wait for a new client:
  EthernetClient client = server.available();
  // when the client sends the first byte, say hello:
  if (client) {
    if (!alreadyConnected) {
      // clear out the input buffer:
      client.flush();
      Serial1.println("We have a new client");
      client.println("Hello, client!");
      alreadyConnected = true;
    }

    if (client.available() > 0) {
      // read the bytes incoming from the client:
      char thisChar = client.read();
      // echo the bytes back to the client:
      server.write(thisChar);
      // echo the bytes to the server as well:
     Serial1.write(thisChar);
   }

  }
}

byte readRegister(byte r)
{
  unsigned char v;
  Wire.beginTransmission(MAC_ADDRESS);
  Wire.write(r);  // Register to read
  Wire.endTransmission();
  Wire.requestFrom(MAC_ADDRESS, 1); // Read a byte
  while(!Wire.available())
  {
    // Wait
  }
  v = Wire.read();
return v;
}
```

JoaquimBoavida, 2021-09-16:
Hi,
Thanks for your help. My CS/SS pin is also PB4  (pin 4 of standard pinout). But I still cannot make it work with "stardard" pinout, only bobuino. For some reason seems to work correctly now, but i'm no sure why.

Even with with your example does not work for me with standard pinout. The eternet chip is not detected. Maybe is a hardware problem. I have more boards. I have to check with other board. Anyway works with bobuino piinout.

Thanks
Joaquim

JoaquimBoavida, 2021-09-17:
Hello,
After some troubleshooting I came to the conclusion that the problem is in my Ethernet modules. They are not wiz850io originals, but clones. I have a defective batch. Several modules are unstable. I use this modules from quite some time and never had a problem, so i didn't think that the problem was there, because I tried several different boards and all have the same problem.
After several hours of troubleshooting without success i decided to build a board from scratch, using a original wiz850io that i have around (the only one!). And everything started to work flawless.
Works with "bobuino" pinout without any software mods and works with the standard pinout  adding the line: 
 
Ethernet.init(4);   

Thank you very much for your support and sorry for all the trouble I caused.
Your work is impressive and appreciated.
Thanks again
Joaquim

closed 2021-09-17 by JoaquimBoavida