MCUdude/MightyCore #224: Adding an USBasp

scrouthtv, 2021-09-26:
Hello,

I recently created a USB to Serial board using a [CY7C65215](https://www.cypress.com/part/cy7c65215-32ltxi). This chip can drive two serial outputs which can be independently configured to UART, SPI and I2C.

I have already confirmed, that the SPI output is working as expected. Now I'd like to use this board for burning the bootloader to an ATmega324PB.

So far, I've set all the fuses, and selected USBasp (MightyCore) from the Programmer dropdown. However, if I press Burn Bootloader, I get this error:

```
avrdude: error: could not find USB device with vid=0x16c0 pid=0x5dc vendor='www.fischl.de' product='USBasp'

avrdude done.  Thank you.

Error while burning the bootloader: Error: 2 UNKNOWN: chip erase error: uploading error: exit status 1
```

It looks like this Programmer is searching for a specific USB to Serial board / chip, which I do not have. Is it possible to modify this Programmer to also use my chip?

Thanks in advance.

MCUdude, 2021-09-26:
Why would you expect that your chip is compatible with the Avrdude USBasp implementation? Does your chip has USB `vid=0x16c0 pid=0x5dc`? If yes, you'll probably have to download [Zadig](https://zadig.akeo.ie) and change from WinUSB to libusb drivers.

scrouthtv, 2021-09-26:
How exactly does the Avrdude USBasp implementation work? I expected to simply be a USB to SPI interface, and the Arduino IDE would do the rest...

The chip's factory programmed to `vid=0x04b4 pid=0x0005`, but I can change it via a configuration utillity. That won't change what the chip does, however.

MCUdude, 2021-09-26:
The USBasp programmer is not a USB to SPI adapter, it is a full-blown ISP programmer designed specifically for programming AVRs. You can't expect that a USB to SPI chip can do the same thing without creating a dedicated implementation for this particular programmer. If you want to know more about the USBasp, you can have a look at its source code.

https://www.fischl.de/usbasp/

scrouthtv, 2021-09-26:
As far as I can tell, it's simply an atmega with the SPI header connected to a header. So with some software, I might be able to program my chip to do the same thing, correct?

closed 2021-09-26 by scrouthtv